ARM-software/CMSIS_4 #2: Change Texas Instruments compiler definition from __TMS470__ to __TI_ARM__

dd430, 2015-11-04:
For all of CMSIS/Include/*.h header files, change (__TMS470__) to new definition for TI ARM Compiler (__TI_ARM__)

For example in https://github.com/ARM-software/CMSIS/blob/master/CMSIS/Include/core_cm4.h 

#elif defined ( __TMS470__ )
  #define __ASM            __asm                                      /*!< asm keyword for TI CCS Compiler */
  #define __STATIC_INLINE  static inline

should be changed to

#elif defined ( __TI_ARM__ )
  #define __ASM            __asm                                      /*!< asm keyword for TI ARM Compiler */
  #define __STATIC_INLINE  static inline

There might be other places in the package where these obsolete definitions are being used.

aethaniel, 2015-11-10:
shouldn't it be 
#elif (defined ( TMS470 ) || defined ( TI_ARM ))
considering your proposal and keep backward compatibility?

ReinhardKeil, 2016-02-29:
We did change the definitions in the upcoming CMSIS drop.  Please review it on:
https://github.com/ARM-software/CMSIS_5/tree/master/CMSIS/Core/Include

closed 2016-02-29 by ReinhardKeil