ARM-software/CMSIS_4 #3: Add '#define __INLINE         inline' definition for TI compiler in core_cm4.h

dd430, 2015-11-04:
Currently (v4.5.0):
#elif defined ( __TMS470__ )
  #define __ASM            __asm                                      /*!< asm keyword for TI CCS Compiler       */
  #define __STATIC_INLINE  static inline

SHould be changed to:

#elif defined ( __TI_ARM__ )
  #define __ASM            __asm                                      /*!< asm keyword for TI ARM Compiler       */
  #define __INLINE         inline                                     /*!< inline keyword for TI ARM Compiler    */      
  #define __STATIC_INLINE  static inline

GuentherMartin, 2016-04-18:
Included in CMSIS-Core.

ReinhardKeil, 2016-09-27:
closing as this is implemented in CMSIS 5 now

closed 2016-09-27 by ReinhardKeil