ARM-software/CMSIS_4 #4: Add TI compiler __SIMD32_TYPE definitions to arm_math.h

dd430, 2015-11-04:
Insert 
#elif defined __TI_ARM__
#define __SIMD32_TYPE int32_t                                        

into this block of code 
#if defined __CC_ARM
#define __SIMD32_TYPE int32_t __packed
#define CMSIS_UNUSED __attribute__((unused))
#elif defined __ICCARM__
#define CMSIS_UNUSED
#define __SIMD32_TYPE int32_t __packed
#elif defined __GNUC__
#define __SIMD32_TYPE int32_t
-------INSERT POINT----------------
#elif defined __TI_ARM__
#define __SIMD32_TYPE int32_t                                        
-------INSERT POINT----------------
#define CMSIS_UNUSED __attribute__((unused))
#elif defined __CSMC__			/* Cosmic */
#define CMSIS_UNUSED
#define __SIMD32_TYPE int32_t
#else
#error Unknown compiler
#endif

GuentherMartin, 2016-04-18:
Included in CMSIS-DSP.

ReinhardKeil, 2016-09-27:
closing as this is fixed in CMSIS 5

closed 2016-09-27 by ReinhardKeil