ARM-software/CMSIS_4 #5: Add TI compiler optimization #if to arm_math.h

dd430, 2015-11-04:
Insert this 

#elif defined(__TI_ARM__)         // Cosmic 

  #define LOW_OPTIMIZATION_ENTER

  #define LOW_OPTIMIZATION_EXIT

  #define IAR_ONLY_LOW_OPTIMIZATION_ENTER

  #define IAR_ONLY_LOW_OPTIMIZATION_EXIT

into this block

#if defined ( __CC_ARM ) //Keil

//Enter low optimization region - place directly above function definition
    #ifdef ARM_MATH_CM4
      #define LOW_OPTIMIZATION_ENTER \
         _Pragma ("push")         \
         _Pragma ("O1")
    #else
      #define LOW_OPTIMIZATION_ENTER 
    #endif

//Exit low optimization region - place directly after end of function definition
    #ifdef ARM_MATH_CM4
      #define LOW_OPTIMIZATION_EXIT \
         _Pragma ("pop")
    #else
      #define LOW_OPTIMIZATION_EXIT  
    #endif

//Enter low optimization region - place directly above function definition
  #define IAR_ONLY_LOW_OPTIMIZATION_ENTER

//Exit low optimization region - place directly after end of function definition
  #define IAR_ONLY_LOW_OPTIMIZATION_EXIT

#elif defined(__ICCARM__) //IAR

//Enter low optimization region - place directly above function definition
    #ifdef ARM_MATH_CM4
      #define LOW_OPTIMIZATION_ENTER \
         _Pragma ("optimize=low")
    #else
      #define LOW_OPTIMIZATION_ENTER   
    #endif

//Exit low optimization region - place directly after end of function definition
  #define LOW_OPTIMIZATION_EXIT

//Enter low optimization region - place directly above function definition
    #ifdef ARM_MATH_CM4
      #define IAR_ONLY_LOW_OPTIMIZATION_ENTER \
         _Pragma ("optimize=low")
    #else
      #define IAR_ONLY_LOW_OPTIMIZATION_ENTER   
    #endif

//Exit low optimization region - place directly after end of function definition
  #define IAR_ONLY_LOW_OPTIMIZATION_EXIT
----------------------------INSERT POINT-------------------------------------
#elif defined(__TI_ARM__)         // Cosmic 

  #define LOW_OPTIMIZATION_ENTER

  #define LOW_OPTIMIZATION_EXIT

  #define IAR_ONLY_LOW_OPTIMIZATION_ENTER

  #define IAR_ONLY_LOW_OPTIMIZATION_EXIT
----------------------------INSERT POINT-------------------------------------
#elif defined(__GNUC__)

  #define LOW_OPTIMIZATION_ENTER __attribute__(( optimize("-O1") ))

  #define LOW_OPTIMIZATION_EXIT

  #define IAR_ONLY_LOW_OPTIMIZATION_ENTER

  #define IAR_ONLY_LOW_OPTIMIZATION_EXIT

…

GuentherMartin, 2016-04-18:
Included in CMSIS-DSP.

ReinhardKeil, 2016-09-27:
Implemented in CMSIS 5

closed 2016-09-27 by ReinhardKeil