ARM-software/CMSIS_4 #6: Licensing information and license text is missing

hugovincent, 2015-11-04:
This repo lacks a license text or other information about the license. While the source files have header blocks with license information, it would be very useful to have a top level LICENSE file that states the license.

ReinhardKeil, 2015-11-05:
I did add a link to the the license document that comes with CMSIS.  
This is also referred by the CMSIS-PACK file that is in the root of the installation.
I hope this is now clearer.

closed 2015-11-05 by ReinhardKeil

hugovincent, 2015-11-05:
Thanks for adding that, but I believe there is still a problem here. The summary says:
> the CMSIS components (CORE, RTOS, DSP, Driver) that are intended to run on the Cortex-M device are today licensed under BSD or ZIP license

but the linked PDF has a wide variety of other terms beyond what the BSD or "ZIP" (presume you mean zlib) license require.