ARM-software/CMSIS_4 #21: CMSIS RTOS Validator: Silence Warnings

ilg-ul, 2016-03-31:
Compiling the files in the Source folder triggered 153 warnings.

Suggested changes:

- fix the code to avoid warnings.

Workaround:

- to silence the warnings, the following GCC options can be use when compiling the files in the Source folder:

```
-Wno-aggregate-return -Wno-conversion -Wno-unused-parameter \
-Wno-missing-prototypes -Wno-missing-declarations -Wno-sign-compare \
-Wno-maybe-uninitialized -Wno-unused-function -Wno-format -Wno-padded
```

Rationale:

- the CMSIS++ code builds with all available warnings enabled.

brondani, 2016-04-15:
It is unclear how to reproduce that as I only get a single warning. However this might be because I'm using new CMSIS-Core files or used a different GCC.

ilg-ul, 2016-04-15:
by default my projects are configured with almost all available warnings, as seen in the attached screenshots.
[image]
[image]
[image]

ilg-ul, 2016-04-18:
could you confirm being able to reproduce the issue, using the attached images?

brondani, 2016-04-18:
Yes, about 150 warnings are triggered when adding the warning options from your attached images, namely:
-Wmissing-prototypes
-Wstrict-prototypes 
-Wbad-function-cast 
-Wunused 
-Wuninitialized 
-Wall 
-Wextra 
-Wmissing-declarations 
-Wconversion 
-Wpointer-arith 
-Wpadded 
-Wshadow 
-Wlogical-op 
-Waggregate-return 
-Wfloat-equal

Anyway if possible please attach your list of warnings, thanks.

ilg-ul, 2016-04-18:
do you really need the actual compiler output? as for the list of distinct warnings, the list of 'counter-measures' I used to avoid your warnings is in the initial ticket text, you can deduce the warnings from there.

ilg-ul, 2016-04-18:
btw, could you make an internal memo so that all code issued by ARM/Keil is also checked in this configuration? (gcc & all those warnings active). 

for running tests, I also suggest using `-Werror`, to break the build on any warning.

brondani, 2016-04-18:
Your compiler output is not necessary, just in case you have it nearby.
And yes I can proceed with an internal memo.
Thank you again for your suggestions.