ARM-software/CMSIS_4 #23: CMSIS RTOS Validator: Check creating multiple instance threads

ilg-ul, 2016-03-31:
The test to create multiple thread instances is not very accurate, it was successful even if my initial code was not correct.

The main problem was that threads were too short, and terminated before trying to create a new instance, so there were not multiple threads in parallel.

Suggested changes:

- add a `osDelay()` in the thread, to make it live longer.

Affected functions:

- `Th_MultiInst()`

brondani, 2016-04-18:
Consider using osKernelInitialize/osKernelStart to gain more control over thread execution.

ilg-ul, 2016-04-18:
can you provide an example how these two functions can help for the Th_MultiInst() test?

brondani, 2016-04-18:
This comment is a reminder to consider using osKernelInitialize/osKernelStart in addition to your suggestion, which is accepted and planned to the next milestone.

ilg-ul, 2016-04-18:
> in addition to your suggestion, which is accepted and planned to the next milestone.

ah, ok, but don't take all my changes as the absolute truth, some where just workarounds, so please analyse them carefully, and if you have a better ideas, feel free to implement them, or to discuss it if you have doubts.