ARM-software/CMSIS_4 #25: CMSIS RTOS Validator: Better control the thread life cycle

ilg-ul, 2016-03-31:
CMSIS allows a thread to terminate, which is a good thing, but does not provide a method for the parent thread to know when a children thread did terminate. (there is no equivalent of the POSIX `pthread_join()`).

This leads to unusual cases, when a thread created by a test terminates during another test.

Please note that calling `osThreadTerminate()` does not guarantee instant termination, some schedulers (like FreeRTOS) will queue the thread on an internal termination list, and the idle thread will perform the actual destruction, possibly much later.

Suggested changes:

- with the current CMSIS RTOS API, this is a bit more difficult, but threads should do their best to ensure that children threads did terminate; at least they should call osThreadTerminate() for all children threads

Affected routines:

- `TC_SignalChildToChild()`

brondani, 2016-04-18:
As a general rule, all test cases shall clean up the resources needed during execution.

ilg-ul, 2016-04-18:
fully agree.