ARM-software/CMSIS_4 #28: CMSIS RTOS Validator: Do not assume immediate reaction for test with interrupts

ilg-ul, 2016-03-31:
The sequences that triggers some actions on interrupts look like this:

```
  NVIC_SetPendingIRQ((IRQn_Type)0);
  ASSERT_TRUE (Stat_Isr == osErrorISR);
```

This might be ok in certain cases, but assumes the interrupt to have top priority and an infinite reaction speed. In my test environment these conditions are not met, and most such tests failed.

Suggested changes:

- One solution is to insert a short delay, to give the CPU and the scheduler enough time to complete the action. This means allowing at leas one scheduler tick, preferably two.

```
  NVIC_SetPendingIRQ((IRQn_Type)0);
  osDelay(2);
  ASSERT_TRUE (Stat_Isr == osErrorISR);
```

Affected routines:

- `TC_GenWaitInterrupts()`
- `TC_MailInterrupts()`
- `TC_MemPoolInterrupts()`
- `TC_MsgQInterrupts()`
- `TC_MutexInterrupts()`
- `TC_SemInterrupts()`
- `TC_SignalInterrupts()`
- `TC_ThreadInterrupts()`
- `TC_TimerInterrupts()`

brondani, 2016-04-19:
Pending interrupts will be executed before the following instruction from the test thread, therefore interrupt priorities should not matter. We have discussed about it internally, not sure that a delay is needed, maybe only barriers instructions. Could you provide more details about your test in this case?

ilg-ul, 2016-04-19:
> priorities should not matter

yes, I was surprised too by the behaviour, but had no time to investigate further.

> maybe only barriers instructions

maybe. 

I would add a separate function to trigger the interrupt (instead of the direct call to `NVIC_SetPendingIRQ()`), and in this single place it will be easier to add either barrier instructions or any other delays, if still needed.

anyway, a short delay there might be useless in some test environments, it might help in others, but I guess it won't be harmful and do not alter the test results in any case.

> more details about your test in this case?

do you have anything specific in mind? I run my tests on a customised version of QEMU and on a STM32F4DISCOVERY board. 

if you implement the single function suggestion, I'll systematically retest with and without delays in both cases and let you know the result, right now there are way to many places to tweak.

---

I reiterate my request to publish the validator as a separate GitHub project, so I can test more easily your changes, right now I have to wait for the next package release for a simple test.

ilg-ul, 2016-04-20:
another possible explanation for some errors might be the lack of `volatile` and high compiler optimisations.

I did not check again your code, but it would be safer to have all variables changed by interrupts as `volatile`.

ilg-ul, 2016-05-06:
Quote from page 331 of "The Definitive Guide to ARM! Cortex!-M3 and Cortex-M4 Processors" by Joseph Yiu:

> Interrupts are imprecise. It means that a number of instructions could be executed after setting the pending status but before the interrupt actually takes place.