ARM-software/CMSIS_4 #31: CMSIS RTOS Validator: Test if a timeout wait really returns before the timeout expires if the event occurs

ilg-ul, 2016-03-31:
The tests do not check if functions waiting on two conditions (event & timeout) really return when the event occurs.

For example a poor timeout implementation would always wait for the entire timeout to expire, and still pass the test, although the behaviour is not correct, the function should return immediately after the event occurs.

Suggested changes:

- time the duration of waiting functions when the event occurs before the timeout. 

Affected routines:

- I added the code to measure the time for `osSignalWait()` in `TC_SignalChildToParent()`, but similar code should be added to all objects

brondani, 2016-04-18:
We should consider to implement it in separate test cases in order to keep the clear purpose of each test case.

ilg-ul, 2016-04-18:
fully agree, one additional test case for each object that implements timed waits.

ilg-ul, 2016-05-02:
To accommodate scheduler frequencies as low as 100 Hz, I changed the logic to perform this test, I triggered the event in the middle of the timeout period (50% out of 100%), and I set the threshold at 75%.

https://github.com/xpacks/arm-cmsis-rtos-validator/commit/1c3584a20e55b4cf983d8ec830ed76d82a184c53