ARM-software/CMSIS_4 #35: CMSIS Documentation: Explain how to create threads with multiple instances

ilg-ul, 2016-03-31:
To my surprise, the feature allowing to define multiple instances of a single thread was completely unknown to me, the current documentation for [threads](https://www.keil.com/pack/doc/CMSIS/RTOS/html/group___c_m_s_i_s___r_t_o_s___thread_mgmt.html) does not explain it at all.

Suggested changes:

- document what the `instances` parameter of the `osThreadDef()` macro is, and how can be multiple instances of a single thread be created.

KeilChris, 2016-04-15:
This is already documented here: http://www.keil.com/pack/doc/CMSIS/RTOS/html/group___c_m_s_i_s___r_t_o_s___thread_mgmt.html
The last bullet point item in the description states that "Active threads can be terminated any time using the function osThreadTerminate. Threads can terminate also by just returning from the thread function."

ilg-ul, 2016-04-15:
done? I updated the suggested changes text, perhaps it was not clear enough.

KeilChris, 2016-04-15:
Will be available in the CMSIS_5 repo within today.

ReinhardKeil, 2017-02-14:
Was RTOS1 API and is in the documentation http://www.keil.com/pack/doc/CMSIS/RTOS/html/group__CMSIS__RTOS__ThreadMgmt.html

closed 2017-02-14 by ReinhardKeil