ARM-software/CMSIS_4 #36: CMSIS RTOS API: Avoid the heavy use of macros (to define objects and to refer to them)

ilg-ul, 2016-04-12:
All `*Def` structures passed to the `*Create()` functions are created using macros, and all these macros, generally used in places where variables are defined, look like function calls, but are not functions, they are structures, making the definitions quite confusing, both for humans and for code analysers.

For example:

```
#include "cmsis_os.h"
 
void Thread_1 (void const *arg);                   // function prototype for Thread_1
osThreadDef (Thread_1, osPriorityNormal, 1, 0);    // define Thread_1
 
int main (void) {
  osThreadId id;
  
  id = osThreadCreate (osThread (Thread_1), NULL); // create the thread
  if (id == NULL) {                                // handle thread creation
    // Failed to create a thread
  }
  
  osThreadTerminate (id);                          // stop the thread
}
```

Suggestion:

Give up using all these macros and find alternate, but more standard, solutions.

For example, [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/) uses the POSIX approach, with optional attributes. It provides both:

- a default, simple, creation method, with minimum arguments and a null pointer to the attributes
- a complete, rich in options, expandable, creation methood, using the attributes.

In CMSIS++, even in the C API, all objects are created with a user supplied storage (one of the structures defined in the `cmsis-plus/rtos/os-c-decls.h`).

For example:

```
...
os_thread_t th1;
os_thread_create(&th1, NULL, func, args);
...
os_thread_destroy(&th1);
...
```

This pattern is consistent across all objects, and matches the use of  the C++ `this` pointer as the first argument to all object methods.

ilg-ul, 2016-04-18:
except for defining constants and driving conditional compilation, the use of macros should be reduced to minimum.

for function like macros, inline functions are preferred.

CMSIS++ makes no use of such macros at all.

ombre5733, 2016-07-01:
Strongly support this issue. I really don't like that one cannot create an array of mutexes or semaphores, for example. Something like the C11-API would be much easier to use. It only specifies that a mutex is represented by a `struct mtx_t`, leaving the content implementation-defined. All functions operate on a pointer of `mtx_t` and do not require macros anywhere.

ReinhardKeil, 2016-09-27:
CMSIS-RTOS2 API has a function interface. Usage of macros is reduced a lot.

ilg-ul, 2016-09-27:
> Usage of macros is reduced a lot.

that's a good thing.

however, at a certain point, you added macros to define attributes. in the current version, if I remember right, these attribute macros vanished, but there are no default, or initialisers, for the attributes, which introduces another issue.

ReinhardKeil, 2017-02-13:
closing as RTOS2 is sloving this

closed 2017-02-13 by ReinhardKeil