ARM-software/CMSIS_4 #37: CMSIS RTOS API: Do **not** make the use of a dynamic allocator mandatory (for stack, queues, etc)

ilg-ul, 2016-04-12:
Although the macros defining objects can be configured to statically allocate storage for objects, some of them internally still allocate storage on the heap, for example for the thread stack.

In CMSIS RTOS running on RTX, I could not find a way to avoid these allocations.

Suggestion:

* add attributes and allow to pass pointers to user defined
storage.

In [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/) all objects can be created with one or more user supplied storages, which can be allocated either statically, or on the stack, or on the heap, according to user needs.

ilg-ul, 2016-04-18:
please note that hiding the dynamic memory allocation behind a pool is also to be avoided, since the size of the pool must be manually configured at compile time.

ReinhardKeil, 2016-09-27:
RTOS2 allows to supply attributes.  Via this structure it is possible to supply static memory. 
For example osThreadNew allows to supply both: static memory for control blocks and static memory for stack.  See:
http://arm-software.github.io/CMSIS_5/RTOS2/html/group__CMSIS__RTOS__ThreadMgmt.html#structosThreadAttr__t

ilg-ul, 2016-09-27:
> RTOS2 allows to supply attributes.

yes, glad you adopted this POSIX solution, as CMSIS++ did originally.

however the documentation on how to do this is incomplete. it states:

> The size of the memory block passed with cb_mem. Must be the size of a thread object or larger.

but I could not find a definition of this size.

also the code to create and use attributes is relatively complicated.

please compare the following use cases from the CMSIS++ C API

```
os_thread_t my_thread;
os_thread_construct(&my_thread, "th1", th1_func, NULL, NULL);
```

with what I guess is the CMSIS RTOS v2 use case, since I could not find an example in the documentation:

```
osThreadAttr_t th1_attr;
memset(th1_attr, 0, sizeof(th1_attr));

th1_attr.name = "th1";

char th1_cb[UNKNOWN_SIZE];
th1_attr.cb_mem = th1_cb;
th1_attr.cb_size = UNKNOWN_SIZE;

osThreadId_t th1_id;
th1_id = osThreadNew(th1_func, NULL, &th1_attr);
```

for your information, the latest µOS++/CMSIS++ C API defines two pairs of functions for creating objects. if the object control block is already allocated, the functions are the equivalent of the C++ constructor/destructor; if the object control block should be allocated, the functions are the equivalent of the new/delete.

for example:

```
// Existing thread storage.
os_thread_t my_thread1;
os_thread_construct(&my_thread1, "th1", th1_func, NULL, NULL);
...
os_thread_destruct(&my_thread1);

// Allocated thread storage.
os_thread_t* my_thread2;
my_thread2  = os_thread_new("th1", th1_func, NULL, NULL);
...
os_thread_delete(&my_thread1);
```

this not only perfectly matches the C++ use case, but also provides a much simpler use case than your.

not to mention that the C++ API is even more easy to use, there is a template that also packs the thread stack in one definition:

```
os::rtos::thread_inclusive<STACK_SIZE> my_thread { "th", th_func, nullptr };
```

jwellbelove, 2017-03-07:
Any use?
http://www.etlcpp.com/