ARM-software/CMSIS_4 #38: CMSIS RTOS API: Add support for critical regions (interrupts & scheduler)

ilg-ul, 2016-04-12:
To my big surprise, the current CMSIS RTOS API specs do not inlude any means to define critical regions.

In [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/), depending on the purpose of the protected region, the user can define regions
where some interrupts are not allowed, or regions where the scheduler context switches are not allowed.

Suggestion:

* define interrupts critical regions (with interupts masked/restored)
* define scheduler criticals regions (with scheduler lock/unlock)

```

  scheduler_status_t
  osSchedulerLock (void);

  void
  osSchedulerUnlock (scheduler_status_t);

  // ...

  interrupts_status_t
  osCriticalEnter (void);

  void
  osCriticalExit (interrupts_status_t);

```

Please note that the only totally safe strategy for implementing nested critical sections is to save the status on enter and restore it on exit. Attempts to use nesting counters are not only more complicated, but do not cover all possible sitations and in some circumstances are even error prone.

ReinhardKeil, 2016-04-14:
osKernelSuspend and osKernelResume implement the osSchedulerLock/Unlock in RTOS2.

ilg-ul, 2016-04-14:
you mean

```
/// Suspend the RTOS Kernel scheduler.
/// \param[in]     tickless      1=tickless, 0=non-tickless
/// \return time in millisec, for how long the system can sleep or power-down.
uint32_t osKernelSuspend (uint32_t tickless);
 
/// Resume the RTOS Kernel scheduler.
/// \param[in]     sleep_time    time in millisec for how long the system was in sleep or power-down mode.
/// \return status code that indicates the execution status of the function.
osStatus osKernelResume (uint32_t sleep_time);
```

but this is a completely different functionality

it looks like you did not grasp the concept of critical sections (or critical regions). this discussion probably belongs to #38.

ilg-ul, 2016-04-14:
in CMSIS++ critical sections are defined in `os-sched.h`

https://github.com/micro-os-plus/cmsis-plus/blob/xpack/include/cmsis-plus/rtos/os-sched.h

the C equivalents are in os-c-api,h, around line 80:

https://github.com/micro-os-plus/cmsis-plus/blob/xpack/include/cmsis-plus/rtos/os-c-api.h

the purpose of critical sections is to protect a sequence of code from interrupts or scheduler.

critical sections may nest (it happens in nested function calls), and this requires a careful implementation, otherwise the protection is terminated prematurely.

the simples and safest implementation requires to preserve the initial status at enter() and to restore it at exit().

critical sections are complemented with uncritical sections, which do exactly the opposite, when running in a critical section allows a sequence of code to run outside the protection condition, and return to whatever status it was before.

CMSIS++ implements both critical and uncritical sections, both for interrupts and scheduler protection.

if you need more details I can provide. 

but critical sections are a... critical (uh!)... aspect of RTOSes, it is very unusual that the CMSIS RTOS API completely ignored them.

as critical, they must be implemented properly, otherwise strange interactions may occur.

0xc0170, 2016-08-19:
> -define interrupts critical regions (with interupts masked/restored)
-define scheduler criticals regions (with scheduler lock/unlock)

+1 for these 2 suggestions. These will require also rethink what to do with RTOS API if interrupts are disabled (many RTOS API currently hardfaults if interrupts are disabled (as in v4.x)).

ilg-ul, 2016-08-19:
> many RTOS API currently hardfaults if interrupts are disabled (as in v4.x)

this is probably specific to RTX.

in the CMSIS++ reference implementation, system calls are perfectly happy to run with the interrupts disabled, since they are plain C/C++ calls, not SVC services.

even if I repeat myself, it is of paramount importance to understand that the only safe implementation is to save the current status at critical_enter() and restore at critical_exit(). avoid any other solutions, like counters, etc since they are not safe.

the current implementation for the interrupts critical section for Cortex-M port is available from:

https://github.com/micro-os-plus/micro-os-plus-iii/blob/xpack/include/cmsis-plus/rtos/port/os-inlines.h