ARM-software/CMSIS_4 #40: CMSIS RTOS API: Add a separate RTC system clock

ilg-ul, 2016-04-12:
The SysTick derived clock is fine for short timeouts, but cannot be used for applications that go into deep sleep (when the SysTick is no longer active) and need to wakeup at specific intervals.

In [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/) there are two system clocks; one is the Systick\_clock, a steady clock that counts ticks, and the Realtime\_clock, an adjustable clock that counts seconds, and can be initially adjusted to reflect the absolute time from the epoch (Jan 1st, 1970) and can also be periodically adjusted using a reference clock.

With multiple clocks, all the objects that define timeouts can be configured to use either the default SysTick clock or the RTC clock, so
the timeout duration depends on the selected clock.

If a hardware RTC is not available, the CMSIS++ RTOS can be configured to derive the seconds from the SysTick clock.

The presence of a RTC clock is consistent with POSIX and ISO requirement to have steady and adjustable clocks.

ReinhardKeil, 2016-09-27:
Not in scope of the RTOS kernel.  That should be really addressed with a CMSIS-Driver.

ilg-ul, 2016-09-27:
the RTC implementation should be done with a CMSIS Driver, but it's use should be defined in the RTOS. otherwise how could you define timeouts based on RTC in system objects?

see the comments in #44.