ARM-software/CMSIS_4 #41: CMSIS RTOS API: Add os_main() to make explicit the use of a main thread

ilg-ul, 2016-04-12:
One of the inovations I appreciated at CMSIS RTOS API was the possibility to make the main() function run as a thread.

However, while working on [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/), I identified two problems with this approach:

- the solutions to implement this feature are not very portable, making almost impossible to run CMSIS test applications in synthetic environments, like POSIX user processes;
- being an optional feature that some CMSIS ports may choose not to implement, the application code is actually more complicated than without this feature, since it must check if the feature is available or not, and possibly start the scheduler, without being able to use other sysm calls in main().

The solution that I used in CMSIS++ was to do not request for this feature  to be implemented at system level, but move it to user space, as an explicit `os_main()` function, always available as a weak function, that runs in the context of a thread started from the standard `main()`.

With this approach, the user is free to choose how it prefers to start the application. If the user wants to have a main that runs in the context of a thread, uses `os_main()`; if not, the standard main() is available, and here the user can handle the scheduler initialisations explicitly, in any custom way.

ilg-ul, 2016-04-18:
the reference implementation for `os_main()` is at:

https://github.com/micro-os-plus/cmsis-plus/blob/xpack/src/cmsis-plus/rtos/os-main.cpp

senselessDev, 2016-09-01:
I think there should be some documentation about how to start the `main()` function as a thread. When using CMSIS RTOS RTX with arm-none-eabi-gcc I was wondering how it will work. I guess, the following code from [RTX_CM_lib.h](https://github.com/ARM-software/CMSIS/blob/62af2d1f68242e73015dfe348ec2a94aefc0faa4/CMSIS/RTOS/RTX/INC/RTX_CM_lib.h#L343) would be the correct one in my case:

```
__attribute__((naked)) void software_init_hook (void) {
  __asm (
    ".syntax unified\n"
    ".thumb\n"
    "movs r0,#0\n"
    "movs r1,#0\n"
    "mov  r4,r0\n"
    "mov  r5,r1\n"
    "ldr  r0,= __libc_fini_array\n"
    "bl   atexit\n"
    "bl   __libc_init_array\n"
    "mov  r0,r4\n"
    "mov  r1,r5\n"
    "bl   osKernelInitialize\n"
    "ldr  r0,=os_thread_def_main\n"
    "movs r1,#0\n"
    "bl   osThreadCreate\n"
    "bl   osKernelStart\n"
    "bl   exit\n"
  );
}
```

Besides, I'm wondering if I'm going to have problems, if I do not use this hook and hence do not have my `main()` running as a thread.

ilg-ul, 2016-09-01:
I updated the link to CMSIS++ os-main.cpp

I still think that a separate os-main() is more portable and easier to use than to play tricks with the startup in order to make main() run on a thread.

senselessDev, 2016-09-01:
> I still think that a separate os-main() is more portable and easier to use than to play tricks with the startup in order to make main() run on a thread.

I agree on that, because portability is better without these tricks. Also, I do not see what advantages  there would be if `main()` is started as a thread?

But besides all that, I am wondering what your role in CMSIS development is? Am I right, that you are dedicated to make CMSIS better in an open source sense but not involved in the KEIL/ARM team? Because you are not satisfied with their work, you did your own C++ implementation (impressive documentation and standard compliance by the way)? And you hope they will use some of your concepts, but it seems that they are not really willing to do it?
I want to get some insights on the different plans here and the future of CMSIS RTOS RTX, to decide whether to use it for future projects or not.

ilg-ul, 2016-09-01:
> I do not see what advantages there would be if main() is started as a thread?

running the application main (`os_main()` in CMSIS++ case) in the context of a thread is only a matter of convenience, the user does not have to handle the details of initialing the scheduler, starting it, etc. and can directly use any system services, for example `osDelay()`. it is very convenient.

> you are dedicated to make CMSIS better in an open source sense

right.

> not involved in the KEIL/ARM team

(unfortunately) not.

> it seems that they are not really willing to do it?

that's up to them to comment on this.

senselessDev, 2016-09-01:
> running the application main (os_main() in CMSIS++ case) in the context of a thread is only a matter of convenience, the user does not have to handle the details of initialing the scheduler, starting it, etc. and can directly use any system services, for example osDelay(). it is very convenient.

Okay, I got the idea, but honestly I prefer the more obvious way, which in my opinion is the explicit starting by the user. Otherwise the level of 'under the hood magic that a beginner does not understand' is getting higher. It is more obvious for me to see the 'kernel' being started. 
So luckily, the result is the same ;) But I think in CMSIS_5 I have read something, that this behavior is changed to our preferred one. 

> that's up to them to comment on this.
Of course, and i see that this repo alone has 67 open comments from you, mostly not answered/discussed. That is the reason for asking myself, how the future of RTX will be. Because Keil sold it for a lot of money back in the day, I thought there is at least one developer continuously working on that. But the repos here does not confirm my thoughts.

ilg-ul, 2016-09-01:
> 67 open comments from you, mostly not answered/discussed.

in a way it is understandable, prior to this version all CMSIS design and development was done behind the closed doors, and the ARM/Keil team has a long way to go to get used with the collaborative spirit of open source development.