ARM-software/CMSIS_4 #42: CMSIS RTOS API: Add support for a synchronised public memory allocator

ilg-ul, 2016-04-12:
Although internally the CMSIS RTOS implementations (like RTX, FreeRTOS) require an allocator for some objects (like thread stacks), the application might also need to allocate memory for specific purposes, and it is not specified how the public allocator should behave, especially how it should handle inter-thread synchronisation.

On top of the standard malloc()/free(), [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/) defines several functions  in the `estd::` namespace, that provide synchronised access to the memory allocator. 

Suggestion:

* add support for synchronised malloc()/free(); syncronisation can be done either by mutex or scheduler critical sections (assuming no allocations will be performed on interrupts).

ReinhardKeil, 2016-04-13:
Memory Pool functions are already public (and thread-safe)
http://www.keil.com/pack/doc/CMSIS/RTOS/html/group___c_m_s_i_s___r_t_o_s___pool_mgmt.html

ilg-ul, 2016-04-13:
negative.

what you are referring to are _memory pools_.

this ticket is about thread-safe malloc()/free(), which is completely different.

https://github.com/micro-os-plus/cmsis-plus/blob/xpack/src/cmsis-plus/libcpp/malloc.cpp

ilg-ul, 2016-04-14:
this ticket too might need some more work to complete.

ReinhardKeil, 2016-04-15:
malloc/free is typically not used in CMSIS-RTOS implementations.  We leave it therefore up to the implementation to solve that issue. 

However, I re-open it as it is a general problem with software libraries that use malloc.  Therefore this issue needs review as we may need to address this in documentation.

For deeply embedded systems it is typically a bad idea to use malloc/free as this may create memory fragmentation. As there is no MMU in Cortex-M, defragmentation is tricky.

ilg-ul, 2016-04-15:
> malloc/free is typically not used in CMSIS-RTOS implementations.

really? where do you get the thread stack in RTX? with CMSIS RTOS API v1, this was one of the items not possible to configure statically.

> We leave it therefore up to the implementation to solve that issue.

this is one possible solution. but for it to work, all allocations must be protected, either with a mutex or even simpler with a scheduler critical section.

> For deeply embedded systems it is typically a bad idea to use malloc/free as this may create memory fragmentation. As there is no MMU in Cortex-M, defragmentation is tricky.

I'm glad you mentioned it.

one of the design requirements of CMSIS++ was to allow fully statically allocated applications. in other words, for all objects it should be possible for the user specify a static storage during object creation.

with CMSIS RTOS v1, this was not possible. I'll check the current v2 and probably create a separate issue for this topic.

ReinhardKeil, 2016-04-15:
RTX uses memory pools.

ilg-ul, 2016-04-15:
> RTX uses memory pools.

this means all threads have the same stack size. and the user must configure at build time the number of pool blocks. I wouldn't qualify this as a flexible solution, but it avoids fragmentation.

ReinhardKeil, 2016-04-15:
RTX therefore also allows user-defined stack in static memory. As said, up-to-now every RTOS has own solutions. 

In ARMv8-M this is even more tricky as, Secure and Non-Secure state have separate thread stacks.

We are not aware of any implementation that solves the malloc/free memory fragmentation problem. If someone is aware of a solution, please let us know.