ARM-software/CMSIS_4 #43: CMSIS RTOS API: Avoid returning agregates (like osEvent)

ilg-ul, 2016-04-12:
Although allowed by the C language, and supported by most compilers, returning structures directly is not a good ideea, since it will trigger lots of warnings in build environments where all warnings are enabled.

Suggestion:

* redesign the API to not return structures, not even structures with bitfields.

0xc0170, 2016-04-12:
It would be useful providing some code snippets or code references as you are referring to specific code within this repository. What kind of warnings do you get? What build environment are you referring to? Is it ``-Waggregate-return`` for GCC which I would assume trigger the warning. Any other toolchain generates similar warnings?

Code snippet for returning osEvent taken from cmsis_os.h in INC folder:

```
/// Get a Message or Wait for a Message from a Queue.
/// \param[in]     queue_id      message queue ID obtained with \ref osMessageCreate.
/// \param[in]     millisec      \ref CMSIS_RTOS_TimeOutValue or 0 in case of no time-out.
/// \return event information that includes status code.
/// \note MUST REMAIN UNCHANGED: \b osMessageGet shall be consistent in every CMSIS-RTOS.
osEvent osMessageGet (osMessageQId queue_id, uint32_t millisec);
```

ilg-ul, 2016-04-12:
> code references

please take a look at the second half of the C wrapper:

https://github.com/micro-os-plus/cmsis-plus/blob/xpack/src/cmsis-plus/rtos/os-c-wrapper.cpp

> What kind of warnings do you get?

in this case aggregate return.

> What build environment are you referring to?

ARM GCC from launchpad

>  Is it -Waggregate-return for GCC which I would assume trigger the warning.

yes

> Any other toolchain generates similar warnings?

I did not test, but I'm pretty sure clang behaves similarly

> Code snippet ... osEvent osMessageGet ()

yes, osEvent is one of the trouble makers, but not the only one, in CMSIS drivers there are many functions returning aggregates.

ReinhardKeil, 2016-04-13:
Agreed, this is addressed in CMSIS-RTOS 2 API.

ReinhardKeil, 2016-04-13:
You made also a comment regarding CMSIS-Drivers.  struct return is used for bit-fields, however this is portable across compilers as long as the bit-field fits into one register.  What problems have you found with that?

ilg-ul, 2016-04-13:
"Although allowed by the C language, and supported by most compilers, returning structures directly is not a good ideea, since it will trigger lots of warnings in build environments where all warnings are enabled." (initial topic)

regardless if it fits or not in a register, returning structs of any size triggers lots of warnings and generally should be avoided.