ARM-software/CMSIS_4 #44: CMSIS RTOS API: Extend the range for osKernelSysTick()

ilg-ul, 2016-04-12:
The 32-bits value of the returned counter allows only small time ranges.

For example, for a 100 MHz CPU, at 1000 ticks/sec, the SysTick counter divisor is 100.000, or 10^5, allowing only 4*10^(9-5), or 40000 ticks, which represent only 40 seconds beween rollovers.

In [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/), the SysTick clock can return separate values in a user provided  structure, allowing any kind of computations related to accurate timings.

Suggestion:

* widen the value to 64-bits.
* return separate ticks & cycles in an optional structure

ReinhardKeil, 2016-04-13:
Addressed with CMSIS RTOS API 2 which adds osKernelTime which returns uint64_t

ReinhardKeil, 2016-04-14:
working with Milestones now

ilg-ul, 2016-04-14:
> working with Milestones now

great!

ReinhardKeil, 2016-09-27:
RTOS2 has a solution for that

closed 2016-09-27 by ReinhardKeil

ilg-ul, 2016-09-27:
> RTOS2 has a solution for that

```
uint64_t osKernelGetTime(void)	
```

this is ok, but the definition is scarce. is this current time relative to the reset moment? what happens with the accuracy if the device goes to sleep?

the theory behind clocks is much more elaborated than you implemented in RTOSv2, a good example is the ISO C++ specs. that's the reason why in CMSIS++ there are several clocks, to cover these practical cases. (some clocks can be adjusted based on an external reference, some cannot; some are relative to reset, some are relative to an epoch; some are expected to work even in sleep, etc)

for your information, in CMSIS++ there are three clocks:
- sysclock, which count scheduler ticks
- rtclock, which counts RTC seconds
- hrclock, with allows to measure high resolution durations

according to POSIX, all functions that handle time points or durations can be configured to use one clock or another, and CMSIS++ allows this, by defining all durations in clock units and adding a clock attribute.