ARM-software/CMSIS_4 #45: CMSIS RTOS API: Use explicit separate calls for different waiting functions

ilg-ul, 2016-04-12:
In POSIX there are three different functions, for example to lock a mutex:

```
  lock();
  try_lock();
  timed_lock();
```

In CMSIS RTOS API all are multiplexed via a single call, the  differentiator being the value of the timeout:

- with 0xFFFFFFFF to mean _wait forever_ (for `lock()`)
- with 0, to mean _try, do not block_ (for `try_lock()`)
- with any other value to mean _wait with timeout_ (for `timed_lock()`)

Although functionally equivalent, I think that the POSIX design increases readability  and this is the pattern used in [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/) for all waiting functions.

Aside from increased readability, another advantage is that some functions like `try_alloc()` can be used in ISRs, while blocking functions like `alloc()` or `timed_alloc()` cannot, and should return an error code.

With the current CMSIS RTOS API, the distinction between function allowed in ISR is not as obvious, for example `osMessageGet()` is specified to be allowed in ISR, but actually it is allowed only with a timeout value of 0.

Suggestion:

* add separate functions like `lock()`, `try_lock()`, `timed_lock()`.

Candidate functions:

* `osSignalWait()`
* `osSemaphoreWait()`
* `osMessagePut()`
* `osMessageGet()`
* `osMailAlloc()`
* `osMailCAlloc()`
* `osMailGet()`
* `osMutexWait()`

ReinhardKeil, 2016-04-14:
I would like to get user feedback here.  To me it looks like we are getting more complex with that modification, especially as we would need to duplicate all the wait variant functions.

ilg-ul, 2016-04-14:
> I would like to get user feedback here

sure, any feedback may be useful, but what more feedback are you expecting from a POSIX standard?

> To me it looks like we are getting more complex with that modification, especially as we would need to duplicate all the wait variant functions.

where do you see this complexity, in the user side or in the implementation side?

because for the user I think that such separate calls obviously increase readability (you can argue, but the meaning of `try_lock()` is by far easier to understand than `lock(0)`) .

as for the implementation side, I just completed the portable implemented for all the CMSIS++ objects starting from scratch, and, believe me, the implementations for the 3 different functions have very little to do with each other; `try_lock()` runs only the test, `lock()` has a loop and in each iteration links the thread to a waiting list, and performs the sleep, while `timed_wait()` does a lot of additional things with the timeouts.

with some extra effort, of course you can implement these three different function in one, but you end up with lots of conditional branches, that need to be evaluated each time at run time.

not to mention that again implementation readability is affected.

you can take a look at the semaphore reference implementation in `os-semaphore.cpp`:

https://github.com/micro-os-plus/cmsis-plus/blob/xpack/src/cmsis-plus/rtos/os-semaphore.cpp