ARM-software/CMSIS_4 #46: CMSIS RTOS API: Make the scheme to assign names to objects more consecvent

ilg-ul, 2016-04-12:
The macros used to define object names are not consecvent, for example the  thread name is derived from the thread function name, but the timer name should be explicitly defined, different from the timer function.

```
void Thread_1 (void const *arg);                          
osThreadDef (Thread_1, osPriorityNormal, 1, 0); 

void Timer1_Callback  (void const *arg);                  
osTimerDef (Timer1, Timer1_Callback);   

```

In [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/), all objects attributes include the `.name` field, making this mechanism  consistent across all objects.

Suggestion:

* use separate explicit names for all objects.

ReinhardKeil, 2017-02-13:
CMSIS-RTOS2 has addressed this

ReinhardKeil, 2017-02-14:
completed

closed 2017-02-14 by ReinhardKeil