ARM-software/CMSIS_4 #48: CMSIS Documentation: Explain that thread functions can return

ilg-ul, 2016-04-12:
The specs currently do not define what happens if a thread exits, although  the validator makes extensive use of thread termination, and thread reuse.

In [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/), a thread can terminate either by returning from the thread  function, or by explicitly calling `Thread::exit()`; alternatively, another thread can use `Thread::kill()` to terminate another thread.

Suggestion:

* add explicit mentions of thread temination, and of thread reuse

KeilChris, 2016-04-15:
This is already documented here: http://www.keil.com/pack/doc/CMSIS/RTOS/html/group___c_m_s_i_s___r_t_o_s___thread_mgmt.html
The last bullet point item in the description states that "Active threads can be terminated any time using the function osThreadTerminate. Threads can terminate also by just returning from the thread function."

ilg-ul, 2016-04-15:
an explicit example would be helpful. if I missed that short phrase (although, believe me, I'm one of the few that took the time to read the documentation), imagine that I'm not the only one.

only when I worked with the validator I clearly got this feature.

ReinhardKeil, 2017-02-13:
closing as there is no further activity

closed 2017-02-13 by ReinhardKeil