ARM-software/CMSIS_4 #49: CMSIS RTOS API: Extend the range of priority levels

ilg-ul, 2016-04-12:
In CMSIS RTOS API there are only 7 thread priorityes, including idle, which might not be enough.

In [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/) the priorities are stored as a byte, and the range (up to 255) can be configured by a user definition.

Suggestion:

* extend the range to 256 levels, and represent it as unsigned (0-255).

Note: apparently addressed in CMSIS 5, but the range is smaller, and still uses enumerated values.

ReinhardKeil, 2016-04-14:
Priority numbers are confusing, especially when you see the NVIC priorities.

ilg-ul, 2016-04-14:
ARM NVIC priorities are indeed very confusing, but here we are discussing thread priorities, which are completely unrelated.

in CMSIS++ the thread priorities have very explicit names, like `thread::priority::normal`, `thread::priority::high`, etc. 

what is confusing in this approach?

ReinhardKeil, 2016-04-14:
The CMSIS++ approach is fine, but pure numbers are confusing, especially when combined with NVIC priorities.

ilg-ul, 2016-04-14:
> but pure numbers are confusing, especially when combined with NVIC priorities.

I have a brain blackout, what pure numbers do you mean? and why do you link them to NVIC priorities?

the actual CMSIS++ thread priorities are defined around line 360 in `os-decls.h`:

https://github.com/micro-os-plus/cmsis-plus/blob/xpack/include/cmsis-plus/rtos/os-decls.h

ReinhardKeil, 2016-04-14:
@ilg-ug CMSIS-RTOS defines a native C API interface, not a C++ API interface.

Confusing is:
In RTOS context, higher values represent typical higher priorities.
In NVIC context, lower values represent higher priorities.

ilg-ul, 2016-04-14:
> @ilg-ug CMSIS-RTOS defines a native C API interface, not a C++ API interface.

CMSIS++ defines both, but the C API is just a wrapper over the C++ API, and this is how it should be, not the other way around, as you try to do.

if you prefer the C API, the priorities are at around line 80:

https://github.com/micro-os-plus/cmsis-plus/blob/xpack/include/cmsis-plus/rtos/os-c-decls.h

> Confusing is:
> In RTOS context, higher values represent typical higher priorities.
> In NVIC context, lower values represent higher priorities.

right. if you do not like the NVIC design, you can complain to ARM ;-)

all variants that we are considering (the ARM CMSIS RTOS API v1, the proposal for v2, the CMSIS++, etc) use higher values for higher thread priorities, so I don't see where the problem is.

ReinhardKeil, 2017-02-13:
CMSIS-RTOS2 has a lot more priority levels.

ReinhardKeil, 2017-02-14:
completed

closed 2017-02-14 by ReinhardKeil