ARM-software/CMSIS_4 #50: CMSIS RTOS API: Add a mechanism to wait for a thread to terminate

ilg-ul, 2016-04-12:
The current specs do not define what happens if a thread exits, and there is no mechanism to tell is a thread has terminated or not.

According to POSIX, in [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/) threads have a `join()` method, that can be used by other threads (usually the parent thread) to wait for thread termination.

Suggestion:

* add `join()`, to allow one thread to wait for a thread to terminate

ombre5733, 2016-07-01:
If `join()` is added, I would also suggest to add `detach()` in order to be able to build a C++11 compatible `thread` class. I have implemented such a thing in [weos](https://github.com/ombre5733/weos/) (see also the [thread-source](https://github.com/ombre5733/weos/blob/master/src/_cmsis_rtos/_thread.hpp)) but would prefer if I did not have to make so many work-arounds.

ReinhardKeil, 2017-02-13:
join attribute has been added to CMSIS-RTOS2

ReinhardKeil, 2017-02-14:
completed

closed 2017-02-14 by ReinhardKeil