ARM-software/CMSIS_4 #52: CMSIS Documentation: Explain the mutex behaviour (recursive vs normal)

ilg-ul, 2016-04-12:
The current [docs](http://www.keil.com/pack/doc/CMSIS/RTOS/html/group___c_m_s_i_s___r_t_o_s___mutex_mgmt.html) do not explicitly specify if the mutex is normal or recursive. However, RTX implements recursive mutexes (only), and the validator expect the mutex to  be recursive.

Suggestion:

* update the documentation to explain the expected mutex behaviour.

ReinhardKeil, 2016-04-14:
almost identical with #53

ilg-ul, 2016-04-14:
not really, issues prefixed with 'CMSIS Documentation' are intended to improve the documentation, which is very sparse in some areas.

#53 is a functional issue, intended to align the mutex behaviour with POSIX mutes specs.

KeilChris, 2016-04-15:
Adding done flag here. Will be available in the CMSIS_5 repo within today.