ARM-software/CMSIS_4 #53: CMSIS RTOS API: Add normal (non-recursive) mutex

ilg-ul, 2016-04-12:
POSIX defines several mutex types, like normal vs recursive.

In [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/), the attributes used to create mutexes include all  POSIX functionality, included type, robustness, priority ceiling, max count.

Suggestion:

* add an explicit method to create normal and recursive mutexes.

ilg-ul, 2016-04-18:
for reference, the CMSIS++ page for mutex is:

http://micro-os-plus.github.io/reference/cmsis-plus/classos_1_1rtos_1_1_mutex.html

the attributes used to construct a POSIX compliant mutex are:

http://micro-os-plus.github.io/reference/cmsis-plus/classos_1_1rtos_1_1mutex_1_1_attributes.html

ombre5733, 2016-07-01:
For reference, [pull-request 15](https://github.com/ARM-software/CMSIS/pull/15) implements such a mutex.

ReinhardKeil, 2016-11-17:
CMSIS-RTOS2 adds attribute for recursive mutex.

ReinhardKeil, 2017-02-13:
closing this now as it has been added in CMSIS-RTOS2 already a while back ago

closed 2017-02-13 by ReinhardKeil