ARM-software/CMSIS_4 #54: CMSIS Documentation: Clarify the specs for binary vs counting semaphores

ilg-ul, 2016-04-12:
The specs mention that creating a semaphore with a count=1 is equivalent to creating a binary semaphore, but expected functionality is not clearly presented.

Suggestion:

* add explicit binary vs counting semaphore definitions, with separate examples.

jkalder, 2016-04-25:
This really needs to be fixed - I would think a binary semaphore is one of the most used feature of any RTOS and it is just NOT available in the Keil ARM CMSIS implementation.
The binary with count = 1 is very misleading. Issuing multiple osSemaphoreRelease() just increments the count irrespective of osFeatureSemaphore value - or is this value hard coded at 65k  in the Keil version?

ilg-ul, 2016-04-25:
if I remember right from my adventure in making CMSIS++ pass the validation suite, the count value is used both as initial count and max count. however this is true in most cases, except when... it isn't (if the initial count is set to 0, the max count is set to a high value).

but the documentation should be updated anyway.