ARM-software/CMSIS_4 #55: CMSIS RTOS API: Allow to explicitly define the semaphore max count

ilg-ul, 2016-04-12:
On one hand, the current specs use the count value both as initial count and as max count, and on the other hand they allow for this count to be 0 (to distinguish between producer and consumer threads).

This is confusing, and leads to unusual code, for example when a max count must be defined, but the semaphore must start with no tokens, a loop of `osSemaphoreWait()` calls must be performed, to return the count to 0.

An explicit max count also has the advantage of making the definitions of binary semaphores obvious (max count is 1).

In [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/), the attributes used for semaphore creation include both an initial and a max count. The default semaphore is created with
the initial count as 0, and MAX_INT for the max count.

Suggestion:

* add an explicit max_count.

ReinhardKeil, 2017-02-13:
CMSIS-RTOS2 has addressed this

ReinhardKeil, 2017-02-14:
completed

closed 2017-02-14 by ReinhardKeil