ARM-software/CMSIS_4 #56: CMSIS RTOS API: Make osSemaphoreWait() return errors, not counts

ilg-ul, 2016-04-12:
`osSemaphoreWait()` returns the semaphore count, no errors like all other calls. At least one CMSIS OS port (the ST one) got this wrong, and returned errors.

In [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/), the `wait()`/`try_wait()`/`timed_wait()` all return error codes, and the semaphore count is obtained by the separate `Semaphore::value()` call.

Suggestion

* change osSemaphoreWait() to return status codes
* add a separate function to return the semaphore count

ilg-ul, 2016-04-18:
for reference, the CMSIS++ documentation for Semaphore is at:

http://micro-os-plus.github.io/reference/cmsis-plus/classos_1_1rtos_1_1_semaphore.html

ilg-ul, 2016-04-18:
according to Scott Meyers, your design breaks "the principle of least astonishment", you should take advantage of what users already know and maximise the likelihood that they guess correctly.

by making most functions return an error, but osSemaphoreWait() return a count, the chances for users to guess correctly without reading the documentation are close to zero.

ReinhardKeil, 2017-02-13:
CMSIS-RTOS2 has addressed this

ReinhardKeil, 2017-02-14:
completed

closed 2017-02-14 by ReinhardKeil