ARM-software/CMSIS_4 #57: CMSIS RTOS API: Add a method to wait for a memory pool block to become available

ilg-ul, 2016-04-12:
The function `osPoolAlloc()` just returns NULL if there are no more available blocks, but does not provide a way to wait for a block.

This design seems fit for use with interrupt routines, where waiting is not an option, but makes things quite difficult when memory blocks are required by a thread.

Also, the design of `osMemoryPool` is not consistent with the design of  `osMailQueue`, which provides blocking calls.

In [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/), the Memory\_pool object is consistent with all other objects, it can be used either in ISR contexts, with `try_alloc()` or in thread contexts, with the blocking `alloc()` and `timed_alloc()`.

Suggestion:

* define all three blocking and non-blocking calls

```
  alloc();
  try_alloc();
  timed_alloc();
```

ReinhardKeil, 2016-04-14:
osPool functions are also used internally by RTOS. We will not combine it with a wait feature.

Question is if we should really add try_alloc and time_alloc as this adds further complexity. Are these functions really needed?  I would like to get user feedback here.

ilg-ul, 2016-04-14:
> osPool functions are also used internally by RTOS. We will not combine it with a wait feature.

??? 

inaccurate, they _may_ be used internally by RTX, but that's completely irrelevant; neither my implementation (µOS++), nor FreeRTOS or µC-OS, use osPool internally.

in my opinion, bending the CMSIS RTOS API specs to match RTX is one of the major weakness of ARM  CMSIS RTOS.

> if we should really add try_alloc and time_alloc

there are several issues here. 

the main one is that your memory pools are not consistent. 
- all other objects block when the resource is not available, only memory pools do not
- even if they also provide a blocking api, other objects can still be invoked from interrupts, if the timeout value is set to 0 

not having a blocking api, memory pools are very hard to use in thread contexts, they probably need to be paired with counting semaphores. 

whether the implementation will use the confusing 3-fold meaning of the timeout parameter, as current CMSIS RTOS does, or use the more explicit and readable POSIX style separate functions, is a different problem, addressed in #45.

ilg-ul, 2016-04-18:
and there is really no additional complexity. the implementation for blocking pool allocations is already available as `osMailAlloc()`. so currently you have two different implementations for block allocations, one is `osPoolAlloc()` which cannot block and the other is `osMailAlloc()`, which can block.

by extending `osPoolAlloc()` with blocking functionality (while still preserving the current functionality),  the #63 issue becomes obvious, mail queues, as separate implementations, are redundant since the  osMail object can be composed from one pool and one queue of pointers (the CMSIS++ reference implementation does it without any problems), so, from the implementation point of view, this suggestion actually reduces complexity, and makes allocation consistent between pools and mail queues.

this also does not impact any internal use of memory pools inside the RTOS, you can continue to call the non blocking version of the `osPoolAlloc()` from interrupts, exactly as you did before.

ombre5733, 2016-07-01:
If [pull-request 14](https://github.com/ARM-software/CMSIS/pull/14) is accepted, one can at least build a blocking memory pool by coupling a memory pool with a semaphore.

ilg-ul, 2016-07-01:
the point is a bit different, it is about consistency and redundancy.

in [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/) I build the mailbox by coupling a blocking memory pool with a queue of pointers.

ReinhardKeil, 2017-02-13:
CMSIS-RTOS2 osMemoryPoolAlloc has this functionality http://www.keil.com/pack/doc/CMSIS/RTOS2/html/group__CMSIS__RTOS__PoolMgmt.html#ga8ead54e99ccb8f112356c88f99d38fbe

ReinhardKeil, 2017-02-14:
completed

closed 2017-02-14 by ReinhardKeil

paolopenna, 2017-03-12:
Please, Let me know:
i understand this is completed but I have a similar issue:
I have a network application on MDK5.21  using "old" method "osPoolAlloc".
This allocation fails when try to send two very close messages.
The network core (the TCP Callback) isn't able to alloc anymore and the full network thread locks in "wait for mutex" state.
Is this the same issue?
Could I solve passing on to the new RTX5?
Thank You very much.

Paolo.