ARM-software/CMSIS_4 #58: CMSIS Documentation: Fix the data type used in osMessageQDef() example

ilg-ul, 2016-04-12:
The current documentation 
[page](http://www.keil.com/pack/doc/CMSIS/RTOS/html/group___c_m_s_i_s___r_t_o_s___message.html)  defines the `type` parameter as _data type of a single message element_, which for message queues should be either int or pointer.

However, the example in `osMessageCreate()` passes a structure, not a pointer to the structure.

```
typedef struct {                                 // Message object structure
  float    voltage;                              // AD result of measured voltage
  float    current;                              // AD result of measured current
  int      counter;                              // A counter value
} T_MEAS;
 
osMessageQDef(MsgBox, 16, T_MEAS);               // Define message queue
```

In the RTX implementation of CMSIS RTOS this works because the data type is ignored by the `osMessageQDef()` macro, being replaced by an `uint32_t` (which, BTW, is not a portable way to store pointers):

```
#define osMessageQDef(name, queue_sz, type)   \
uint32_t os_messageQ_q_##name[4+(queue_sz)] = { 0 }; \
const osMessageQDef_t os_messageQ_def_##name = \
{ (queue_sz), (os_messageQ_q_##name) }
```

KeilChris, 2016-04-15:
Adding done flag here. Will be available in the CMSIS_5 repo within today.

ReinhardKeil, 2016-09-27:
Fixed in CMSIS-RTOS5 as discussed

closed 2016-09-27 by ReinhardKeil