ARM-software/CMSIS_4 #62: CMSIS RTOS API: Fix osMessagePut()/osMailPut() inconsecvent error when called from ISR

ilg-ul, 2016-04-12:
Most other calls return `osErrorISR` when called from ISR.

`osMessagePut()` (and its sister `osMailPut()`), when called from ISR with a timeout different from 0, returns `osErrorParameter`, as if the problem would be the parameter, not the context where it is called from.

In [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/) this problem is simply avoided by having distinct calls, with all blocking calls banned from ISR, and all `try_xxx()` calls allowed from ISRs.

ReinhardKeil, 2016-04-14:
Related to POSIX error codes

ilg-ul, 2016-04-14:
that's correct, once accepting the use of POSIX codes, the current inconsequent behaviour would no longer be an issue.