ARM-software/CMSIS_4 #63: CMSIS RTOS API: Mail queues, as separate objects, are redundant

ilg-ul, 2016-04-12:
A mail queue is a combination of a memory pool and a message  queue, with very similar usage, which makes it redundant.

Actually in [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/), with carefully designed C++ objects, it was perfectly possible to implement the `osMailQ` by composing it from a pool and a queue.

```
#define osMailQStaticDef(name, items, type) \
struct { \
    osMailQ data; \
    struct { \
      type pool[items]; \
    } pool_storage; \
    struct { \
      void* queue[items]; \
      os_mqueue_index_t links[2 * items]; \
      os_mqueue_prio_t prios[items]; \
    } queue_storage; \
} os_mailQ_##name; \
const osMailQDef_t os_mailQ_def_##name = { \
    #name, \
    (items), \
    sizeof (type), \
    sizeof (void*), \
    &os_mailQ_##name.pool_storage, \
    sizeof(os_mailQ_##name.pool_storage), \
    &os_mailQ_##name.queue_storage, \
    sizeof(os_mailQ_##name.queue_storage), \
    &os_mailQ_##name.data \
}
```

ReinhardKeil, 2016-04-14:
This may contribute to additional complexity.

ilg-ul, 2016-04-14:
same question as for #45, where do you see this complexity, in the user side or in the implementation side?

to summarise, the current solution includes:
- a memory pool that is not consistent with other objects, and requires a semaphore when using in threads
- a message queue that is not able to pass messages, but only pointers
- a mail queue, which is not consistent with the memory pool, since it has blocking allocations, and uses the crippled message queue to pass pointers to consumer threads.

with the current solution:

- if you need a standard POSIX queue, which passes more than pointers, you need to use a mail queue, and manually handle allocation and deallocation, instead of the very simple two way copy used by a standard queue.

- if you need only a blocking memory pool, and prefer to handle the queue separately (for example because CMSIS queues do not handle priorities), again you cannot use the memory pool, but need to use a mail queue and throw away the queue functionality.

does this look very simple to you? 

you may be very well used to this approach due to your intimate knowledge of RTX, and do not notice the problems, but for regular users this is a nightmare.

as for the CMSIS++ approach, it might not be perfect (suggestions welcomed!), but it is simple and straightforward:

- there is a POSIX compliant queue, with messages of any size and priorities, which can be used either blocking from threads or non-blocking from interrupts (the `try_send()` and `try_receive()` variants)

- there is a Memory_pool object, with consistent API, which can be consistently used either blocking from threads or non-blocking from interrupts (the `try_alloc()` variant)

if the user decides the 2-way copy involved by the queue is too much (for large buffers, for example), it can obviously pair a Memory_pool with a queue and pass only pointer to buffers, but it can also do more, it can also pass a small structure that includes the pointer, and also be able to add additional information that was not available at the moment when the buffer was filled in.

in my opinion, the explicit use of a memory pool and a queue of custom structures is preferable to the current CMSIS RTOS MailQ objects.

as a side note, as I already mentioned, in CMSIS++ the ARM CMSIS RTOS compatibility wrapper implements the MailQ using the CMSIS++ memory pool and message queue objects without any problems, the validator passes happily.

for reference, the Memory_pool API is documented at:

http://micro-os-plus.github.io/reference/cmsis-plus/classos_1_1rtos_1_1_memory__pool.html

and the Message_queue API is documented at:

http://micro-os-plus.github.io/reference/cmsis-plus/classos_1_1rtos_1_1_message__queue.html

ReinhardKeil, 2017-02-13:
closing as it appears to be a duplicate

closed 2017-02-13 by ReinhardKeil