ARM-software/CMSIS_4 #65: CMSIS RTOS API: Use POSIX error codes

ilg-ul, 2016-04-13:
The CMSIS RTOS API defines a set of error codes, but the semantic is proprietary (and, to my oppinion, not very consistent).

Based on POSIX, [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/) uses the standard POSIX error codes from the <errno.h> header file, the only specific definition being `result::ok` to indicate that no error occured.

Suggestion:

* use the standard POSIX error codes instead of reinventing new codes.

ReinhardKeil, 2016-04-13:
Are these the POSIX error codes you refer too?
http://dox.ipxe.org/group__posixerrors.html#ga7a97ba3740a8a51d0f6f5d30b1b7906d

ilg-ul, 2016-04-13:
The current error codes are listed in the online POSIX web at:

http://pubs.opengroup.org/onlinepubs/9699919799/functions/V2_chap02.html#tag_15_03

The actual numbers behind the codes are defined in the <errno.h> or <cerrno> headers.

ilg-ul, 2016-04-13:
But don't be intimidated by the long list, in CMSIS++ I used only a subset of them. I don't have a list of them, since it wasn't necessary, but you can see them if you browse the reference pages:

http://micro-os-plus.github.io/reference/cmsis-plus/group__cmsis-plus-rtos.html

one example might be the [Semaphore::timed_wait()](http://micro-os-plus.github.io/reference/cmsis-plus/classos_1_1rtos_1_1_semaphore.html#a750ddba1b73e8ecf04f175de8d21b83b) call

ilg-ul, 2016-04-18:
I did a consistency check in CMSIS++ and now all functions like `try_lock()` return `EWOULDBLOCK` if the resource is busy.

the list of POSIX errors currently in use is available from:

http://micro-os-plus.github.io/reference/cmsis-plus/namespaceos_1_1rtos_1_1result.html

ombre5733, 2016-07-01:
I strongly support this proposal. Having to deal with fewer different definitions of error codes makes it much easier for the developer. Plus POSIX is really wide-spread and well-known.

ReinhardKeil, 2016-11-17:
We have decided not to go for a POSIX compliant implementation

ReinhardKeil, 2017-02-13:
closing as there is no further activity

closed 2017-02-13 by ReinhardKeil