ARM-software/CMSIS_4 #66: CMSIS RTOS API: Add typedefs for all different types used in prototypes

ilg-ul, 2016-04-13:
It is a good programming practice to define separate types for each different function parameter, instead of using generic `uint32_t`; done properly this usually increases readability.

In [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/) all types have separate definitions (for example: `clock::duration_t`, `scheduler::status_t`, `flags::mask_t`, `thread::priority_t`, and so on).

Suggestion:

* add typedefs for all different types used in prototypes

ReinhardKeil, 2016-10-14:
I think is is now completed within reason.

ilg-ul, 2016-10-19:
as far as I can see, the `cmsis_os2.h` still has lots of plain `uint32_t`. consider something like `duration_t`, `flags_t`, `size_t` (from `stddef.h`), etc.