ARM-software/CMSIS_4 #67: CMSIS RTOS API: For all objects, add reset functions to return the object to initial status

ilg-ul, 2016-04-13:
In [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/) objects can be returned to the inital status with `reset()`.

Suggestion:

* add reset() to the queue object
* add reset() to the memory pool object
* add reset() to the mail object
* add reset() to the semaphore object
* add reset() to the mutex object

ombre5733, 2016-07-01:
Isn't this a dangerous functionality? I cannot imagine, when to reset a mutex for example. A (plain) mutex has two states: locked and unlocked. If a thread locks a mutex, that very thread has to unlock it again. Calling `mutex::reset()` from another thread does not seem like a good idea because that would allow access to resources which are not available.

Especially in the C++ context I do not see the need of `reset()` functions immediately. There one should use RAII classes like `std::lock_guard` which should prevent misusage (such as forgetting to unlock a mutex).

ilg-ul, 2016-07-01:
> I cannot imagine, when to reset a mutex for example

imagine you have the object created by a static constructor, you detect a problem that is not catastrophic to require a total reset, and you need to perform only a partial reset of the affected part of the application.

> Isn't this a dangerous functionality?

it is as dangerous as having a NVIC Reset.

ombre5733, 2016-07-01:
Ok, let's assume that a problem has been detected. Why does the thread which holds the mutex not release it? Allowing another thread to release the mutex does not seem like a good idiom to me. How would the other thread know, that it is safe to release the mutex, for example.

ilg-ul, 2016-07-01:
when the monitoring thread detects a problem, all affected threads are interrupted/cancelled, and everything starts afresh. I think it is easier to reset all common global synchronisation objects than expect each individual thread to perform various cleanups.

ReinhardKeil, 2016-10-14:
Are the osXxxxxDelete functions not good enough to recover from catastrophic situations?  When you want to restart several software components, you can simply create the objects again.

ilg-ul, 2016-10-14:
actually this is not necessarily related to catastrophic situations.

in the latest version of CMSIS++, in the POSIX I/O subsystem, I have a class that implements a POSIX device character, which uses several semaphores. each time I open the device I need to reset() all semaphores.

> you can simply create the objects again.

given your use of dynamic memory for creating objects, this is less desired, for multiple reasons.

ReinhardKeil, 2017-02-13:
There seems to be no demand for this functionality