ARM-software/CMSIS_4 #69: CMSIS RTOS API: For memory pools, add more accessors to get pool status

ilg-ul, 2016-04-13:
In CMSIS RTOS API, once created, a memory pool has no functions to tell if the pool is empty or full, to get the block size, the pool capacity, and so on.

In [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/) all memory pool characteristics are available via accessor functions.

```
std::size_t
capacity (void) const;

std::size_t
count (void) const;

std::size_t
block_size (void) const;

bool
empty (void) const;

bool
full (void) const;
```

Suggestion:

* add accessor functions for memory pool status

ReinhardKeil, 2016-11-17:
CMSIS-RTOS2 adds the requested features
http://arm-software.github.io/CMSIS_5/RTOS2/html/group__CMSIS__RTOS__PoolMgmt.html

ReinhardKeil, 2017-02-13:
closing as there is no further activity

closed 2017-02-13 by ReinhardKeil