ARM-software/CMSIS_4 #70: CMSIS RTOS API: For message queues, make the message size user configurable

ilg-ul, 2016-04-13:
The current queue specifications limits the message size to an integer (and RTX internally uses a void*).

Based on POSIX, [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/) does not limit the message size; instead it is configured at queue creation.

Suggestion:

* extend the queue creation API and add the message size

ReinhardKeil, 2016-04-15:
we would need a solution to prevent from memory fragmentation

ilg-ul, 2016-04-15:
I'll rephrase, since my request was not clear enough.

for a POSIX queue the message size can be configured when the queue is created, and all messages share this size as max size. so no fragmentation occur.

in CMSIS RTOS v1, this size is limited to 4 bytes, and I could not find a way to change it.

in CMSIS++, the message size is passed when the queue is created:

http://micro-os-plus.github.io/reference/cmsis-plus/classos_1_1rtos_1_1_message__queue.html

ReinhardKeil, 2016-04-15:
looks like the feature is then similar to osMail http://www.keil.com/pack/doc/CMSIS/RTOS/html/group___c_m_s_i_s___r_t_o_s___mail.html

ilg-ul, 2016-04-15:
it isn't. perhaps a careful read of #63 will clarify this.

ReinhardKeil, 2016-04-15:
similar != identical

osMail does not copy information.  It provides a buffer that can be filled and passed to a consumer.

ilg-ul, 2016-04-15:
POSIX queues are a more general (and standard) concept. applications that are happy with the 2-ways copy should be allowed to use them, especially since the API is simple (send()/receive()), and not be forced onto more complicated APIs that require additional explicit alloc()/free() for each message.

as explained in #63, osMail is an object composed of a memory pool and a queue of pointers, not the other way around.

ilg-ul, 2016-04-15:
I just reviewed the proposed `osMessageQueueNew()`. since this is the place where the message size parameter should be defined, adding it later will require another creation function, or breaking compatibility.