ARM-software/CMSIS_4 #72: CMSIS RTOS API: For message queues, add message priorities

ilg-ul, 2016-04-13:
The current queue specifications do not mention anything about message priorities, apparently all messages have equal priority and the queue policy seems to be FIFO.

Based on POSIX, in [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/) message queues have priorities, and messages are retrieved in priority order.

Suggestion:

* extend the API and add priorities to the functions storing/retrieving messages from the queue

ReinhardKeil, 2016-04-15:
at this moment, it is out of scope.  However, I consider it as a good suggestion for future improvements, hence labeled as 'future'.

ilg-ul, 2016-04-15:
please be aware that adding priorities later will require to add either new put()/get() functions, or new parameters the existing ones, breaking compatibility.

ilg-ul, 2016-04-18:
for reference, the CMSIS++ documentation for message queus is at:

http://micro-os-plus.github.io/reference/cmsis-plus/classos_1_1rtos_1_1_message__queue.html

ReinhardKeil, 2017-02-13:
Added to osMessageQueuePut

ReinhardKeil, 2017-02-14:
completed

closed 2017-02-14 by ReinhardKeil