ARM-software/CMSIS_4 #73: GitHub workflow: use of separate development repositories not scalable

ilg-ul, 2016-04-13:
The idea to bring the CMSIS repository to GitHub is great, but the recent idea to create a separate repository for the development version of CMSIS_5 doesn't fit very well in the usual GitHub workflow and also doesn't scale well. Will you open a new development repository for each future version you will work on? What will you do when the CMSIS 5 release is out? Remove this repository? (and loose all the commit and issue history).

The usual approach is to use Git branches and tags.

Normally one single development branch (let's call it `devel`, but any name you like is fine) should be enough. 

The `master` branch remains stable at the latest publicly released version, and all new commits go to the `devel` branch. When the new release is ready, you merge the `devel` commits into `master`, add a new release tag, and restart the process.

If you look far ahead into the future and start work on a new version before the current one is released, multiple branches can be created, and merged from time to time, as needed, Git (and GitHub) do not have limitations in this respect.

ReinhardKeil, 2016-04-13:
We are still exploring the best practice with GitHub.  The problem with the CMSIS repository was that it contained the build version rather than the development version. CMSIS_5 is the development version and we have just introduced branches.  We believe that Branches will allow use to keep just one GitHub repository in future for development.

closed 2016-04-13 by ReinhardKeil

ilg-ul, 2016-04-14:
I checked and you added a `develop` branch to the CMSIS_5 repository, only to increase confusion where the active development branch is. it makes little sense to have a `develop` branch in a separate development repository, you can safely use the `master` branch there, since there is no confusion, the stable files are safe in the CMSIS repository.

in my opinion, there should be no distinction between the build and the development repository. 

based on the common workflow, the expected location of the `develop` branch is in the main `ARM-software/CMSIS`; the CMSIS_5 repository is a small anomaly that sooner or later should be abandoned (the later, the more problems you'll have, and more project history you might loose).

you can easily migrate the development files from CMSIS_5 to CMSIS at any time; create the `develop` branch in CMSIS, copy all files from the active branch of CMSIS_5 (master or develop, I don't know which one is now) to CMSIS, commit, push the changes to GitHub and, from this moment on, ask your colleagues to use the final CMSIS repository.

this manual file copy will preserve the latest versions of the files, but will loose the CMSIS_5 commit history.

if you want to preserve this history (you have 78 commits right now), you can do this, but it requires some more advanced Git handling, and I'm not sure you can manage it correctly from the very beginning. it involves to configure your local version of the CMSIS repository with a reference to the CMSIS_5 repository, and in the CMSIS/develop branch to perform a merge of the CMSIS_5 active branch. (if you break anything, as long as you do not push the changes to GitHub, it is not a problem, you can always remove the local broken repository and repeat the procedure).

I don't know what are your preferred Git tools, but the GitHub Desktop is not featured enough for this kind of operations; I suggest you use [Source Tree](https://www.sourcetreeapp.com) from Atlassian.

returning to the workflow, when you decide the release is ready, do a check & cleanup in the `develop` branch, go to the `CMSIS/master` branch and do a merge with the `develop` branch.

then run your scripts to prepare the CMSIS Pack archive, and upload the two files in the Releases section and obviously copy them to your Packs server.

hope this helps.

(this issue, closed or not, will probably require some further work to complete)

ReinhardKeil, 2016-04-15:
Thanks for the input.  I agree that needs further work.

However we are not going to use https://github.com/ARM-software/CMSIS for development.
To align with our internal processes we are using use https://github.com/ARM-software/CMSIS_5 for development.  Give us a few days to resolve that, and if still has issues you should open a ticket on CMSIS_5.

ilg-ul, 2016-04-19:
I noticed all code contributions to be a pull request immediately followed by a merge, apparently executed by the same person.

If my observations are right, any particular reason for this unusual commit pattern?

Team members with write access to the repository can directly commit to GitHub. Pull requests are intended to third parties without special access to the repository, and for code reviews.