ARM-software/CMSIS_4 #74: GitHub Issues workflow

ilg-ul, 2016-04-13:
Since you mentioned you are still exploring the best practice with GitHub, please also consider some suggestions to help you improve your Issues workflow.

When you receive a ticket, it is polite to acknowledge it by tagging it with one of the available **Labels** (bug, enhancement, question, etc), and possibly assign it to the team member responsible either to implement it or to post comments in the ticket (if the team member is not public).

Then, for issues that are accepted, it is useful to assign a milestone. For example now you can define a milestone named _CMSIS 5_ and tag all issues that will be included in CMSIS 5 with it.

Note: you can create any custom labels, as well as any milestones, with the GitHub interface (the two buttons, Labels and Milestones, in the Issues page).

It is polite for contributors to close their own issues when they are comfortable with the outcome, but it is not uncommon for them to forget, and in this case solved issues are usually closed by the maintainers after each release.

Unless reports are completely off-topic, it is not particularly polite to unilaterally close issues without a reaction from the contributor, because usually you'll continue to receive comments even for closed tickets. In these cases, for the issues that were not actually solved, it is usually polite to reopen the tickets.

As for any collaborative environment, it is useful to discuss the possible solutions with the contributors, since from these discussions usually better solutions are identified. 

Replies like "We will investigate and integrate it (if applicable) into the next release", are not particularly useful.

In the GitHub workflow, there are two kind of contributions, **pull requests**, for which the contributors provide actual code that, if accepted, can be directly merged into one of the branches, and **issues** when the contributor provides only a description of the problem and you have to work on the code.

Both share a common database, and usually it is redundant to register both an issues ticket and a pull request.

However it is common for discussion in the Issues page to reach some kind of conclusion, and the contributor to be asked to submit a pull request with the changes.

I know that large corporations have strict internal processes and sometimes it is difficult to embrace the open source collaborative culture, but since you already did the first big step, to move the CMSIS development to GitHub, you're on the right path :-)

(you just need to get used to the idea, and finally accept to move the development for all the other CMSIS packs on GitHub).

ReinhardKeil, 2016-04-14:
Thanks for these hints.  Very helpful.  Will try to follow them and educate the team.

ilg-ul, 2016-04-14:
you're welcome! and good luck!

ReinhardKeil, 2016-11-17:
@ilg-ul Does it work better now? How can we further improve?

ilg-ul, 2016-11-17:
I had difficulties to understand what the milestones were, if they were reached, if there were releases, if these releases were candidates or final, if the documentation reflects the code, things like these.

but go ahead, bring other projects to GitHub, like the validator, and the workflow will improve with time.