ARM-software/CMSIS_4 #80: CMSIS RTOS API v2: Fix inconsistent type naming convention

ilg-ul, 2016-04-15:
This is a recuring problem, that seems to affect all CMSIS components, not only RTOS, and, if I remember right, I already reported, but apparently without effect.

In the current `cmsis_os.h` I can read the following types:

* `osThreadState`
* `os_timer_type`
* `osThreadAttr_t`

In other words you freely mess camel case names with underscores, with or without the standard `_t` suffix.

Even worse, types are not consisten even inside a single definition (the struct name uses one naming convention and the typedef another naming convention):

```
typedef struct os_mutex_attr {
  const char                 *name;    
  uint32_t               attr_bits;    
  uint32_t                reserved;    
} osMutexAttr_t;

```

In my oppinion your general naming convention, not only the type naming convention, is problematic, but the issue here is not which naming convention you choose (this may turn into a religious war), but once you choose it, stick to it consistently.

[CMSIS++](http://micro-os-plus.github.io/cmsis-plus/) uses the lower case naming convention, also adopted by POSIX and by the ISO C/C++ standards, with types suffixed by `_t`. User class names may have the initial character in uppercase.

Suggestion

* make the type naming convention uniform, preferably using the POSIX convention, with lower case and the `_t` suffix.

ReinhardKeil, 2016-04-15:
Agreed

ilg-ul, 2016-09-15:
The new `cmsis_os2.h` seems to consistently use the`_t` suffixes for types and `_s` for structures, so I guess this issue can be closed.

closed 2016-09-15 by ilg-ul