ARM-software/CMSIS_4 #81: CMSIS RTOS API v2: osEventFlagsWait()/osThreadFlagsWait() return type inconsistent with other blocking calls

ilg-ul, 2016-04-15:
Most blocking calls return a status code, usually ok or timeout.

It is not clear how you can differentiate the ok from timeout, in case of the flags functions, but it is inconsistent.

In [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/), all blocking functions return a consistent status code, and the output flags are passed via a pointer parameter.

http://micro-os-plus.github.io/reference/cmsis-plus/classos_1_1rtos_1_1_event__flags.html

Suggestion:

- change the return type to a status code and add a pointer parameter for the output flags

ReinhardKeil, 2016-04-15:
This is not part of CMSIS v4 -> please use https://github.com/ARM-software/CMSIS_5

closed 2016-04-15 by ReinhardKeil

ilg-ul, 2016-04-15:
sorry, I'm not contributing to a repository whose faith is unclear and might be removed at any time to make place for CMSIS_6, CMSIS_7, etc

ReinhardKeil, 2016-04-15:
OK, we will position CMSIS_5 then correctly.

ilg-ul, 2016-04-15:
do as you wish, but having two repositories for the same project is meaningless and counterproductive.