ARM-software/CMSIS_4 #83: RTE_Components.h missing

hackrid, 2016-04-26:
Hi,

I want to use the existing CMSIS-Drivers in a project.
Compilation fails, telling me that RTE_Components.h is missing.

where can I get this file? could you post an example?

hackrid, 2016-04-26:
ok, I found something...
https://github.com/ARM-software/CMSIS/blob/master/CMSIS/DAP/Firmware/Examples/LPC-Link-II/RTE/RTE_Components.h

Seems CMSIS-Driver is not intended to be used as stand-alone solution?

jkrech, 2016-04-26:
RTE_Components.h is generated as part of the Runtime configuration.
For each selected component the text between <RTE_Components_h> </RTE_Components_h> is collected in the RTE_Components.h file

`<component Cclass=...>`
  `  <description> ...  </description>`
  `  <RTE_Components_h>`
    `    #define MY_COMPONENT`
  `  </RTE_Components_h>`
  `   ...`
  `</components>`

This allows components to check what other components are active in a project.

hackrid, 2016-04-26:
generated from what?

My intention is to use the CMSIS-Drivers with my own buildsystem

ilg-ul, 2016-04-26:
> generated from what?

generated by MDK

> My intention is to use the CMSIS-Drivers with my own buildsystem

good luck with that!

for the moment the whole story with CMSIS Packs, Drivers, RTOS API is just a marketing gimmick to promote ARM/Keil tools.

about two years ago I implemented a CMSIS Packs manager in GNU ARM Eclipse hoping that I'll be able to use the content of the packs to create my own projects. no way, all examples are MDK specific, and most configurations are ARM CC specific. so, the only really portable content from the packs are the SVD files (incomplete, as discussed in another ticket) that I use in the debug peripheral view. the rest of the packs content is largely unusable.

hackrid, 2016-04-26:
:'(

jkrech, 2016-04-27:
You can find CMSIS-PACK eclipse plug-ins and source code here 
https://github.com/ARM-software/cmsis-pack-eclipse
This provides you a framework that allows you to configure your software and as part of the project update also the RTE_Components.h is generated.

ilg-ul, 2016-04-27:
@hackrid,  let me know if you managed to create a useful, functional project with those plug-ins. on my development platform (OS X) all attempts failed with errors/exceptions, and on Linux some of the views were not displayed properly. on Windows I do not know (neither I'm very interested).

hackrid, 2016-04-27:
@ilg-ul I don't really want to use this plugin since you mention it is buggy...
My intention for now is to hand-write the RTE_*.h files to get the CMSIS-Driver working with my existing code. hope this is feasible.
I have very little interest in the Windows platform, too.

@jkrech what is your oppinion regarding this plugin? should it work?

ReinhardKeil, 2016-06-13:
The generation of the RTE_Components.h file is described here:
http://www.keil.com/pack/doc/CMSIS/Pack/html/pdsc_components_pg.html#RTE_Components_h

The plugin's are designed to be platform independent and should also work on Linux and OS X.

ReinhardKeil, 2016-09-27:
Closing as there is no further activity

closed 2016-09-27 by ReinhardKeil