ARM-software/CMSIS_4 #89: CMSIS Drivers: Instance parameters needed

0x8000-0000, 2016-05-15:
I find the API for the CMSIS drivers, specifically the USART and SPI confusing. They seem to assume there is only one instance of USART or SPI peripheral in use, or that different peripherals will have different function implementations.

For instance, this is the ARM_USART_Send signature:

      int32_t ARM_USART_Send (const void *data, uint32_t num);

If my application / board uses both USART1 and USART2, how do I differentiate between the two?

I understand that I could write ARM_USART_Send_using_USART1 function and ARM_USART_Send_using_USART2 function, and populate each in the respective ARM_DRIVER_USART but that seems quite non-intuitive, especially when on a piece of silicon all similar functional blocks have similar if not identical capabilities, down to the register interface. Ignoring that USART1 runs on a faster bus than USART2 and might need slightly different initialization sequences, the read and the write are identical (save for the base address of the register set).

ReinhardKeil, 2016-06-13:
Please take a look here where this is explained:
http://www.keil.com/pack/doc/CMSIS/Driver/html/_theory_operation.html#DriverInstances

ilg-ul, 2016-06-13:
see also #86

0x8000-0000, 2016-06-13:
@ReinhardKeil "data structure with pointers to implementation functions" is fine to represent device / bus instances, but what about the actual _implementation_ ? For the chips that I'm working with (STM32, Kinetis Kxy, Energy Micro), there is no difference in the implementation between the function that sends data through USART2 and the function that sends data through USART1.  They use similar registers.

What you are saying here is, if my application uses 3 USARTs I should write and test three functions? And they will all use up some flash space?

ilg-ul, 2016-06-13:
@0x8000-0000, if you take a look at how the current drivers are implemented, you'll discover that each instance of a driver has a separate array of pointers to separate functions.

yes, if your device has 8 USARTs, the CMSIS Drivers include 8 separate sets of implementation functions! internally each such set refers to a separate RAM area, so it is possible for all 8 sets to share a common function, but this is an implementation choice.

to be sure you got it right, for the 8 USART case, there are 8 Initialise functions (USART1_Initialize(), USART2_Initialize(),...), 8 send functions, 8 receive functions, and so on.

each such function is implemented like this:

```
static int32_t                 
USART1_Initialize (ARM_USART_SignalEvent_t cb_event)                   
{ 
  return USART_Initialize (cb_event, &USART1_Resources); 
}

static int32_t                 
USART2_Initialize (ARM_USART_SignalEvent_t cb_event)                   
{ 
  return USART_Initialize (cb_event, &USART2_Resources); 
}
```

so deep inside the implementation there is an instance specific area, but access to it it is indirect.

this approach is totally unusual in the object oriented world, where class member functions have a first parameter to the instance data.

0x8000-0000, 2016-06-14:
@ReinhardKeil the "WONTFIX" label seems more appropriate.

0x8000-0000, 2016-06-14:
@ilg-ul Yeah, I see... but why?

I understand this change has some ripple effects, but due to issues such as this one, I doubt it is in widespread commercial use.

ilg-ul, 2016-06-14:
> due to issues such as this one, I doubt it is in widespread commercial use

the idea of common drivers is generous. unfortunately the current design is not very inspired (to put it mildly) and the implementation is MDK centric. so yes, I also doubt it'll be successful.

another problem I encountered was the non-reentrant callbacks. (try to use the same callback in two instances of the same driver and you'll see what I mean).

my view on this issues is to design the objects in C++, then add C APIs on top of the C++ objects. this keeps the design clean and allows a convenient C++ and C integration.

I did some experiments with C++ drivers (serial and usb) and I plan to further develop them in CMSIS++.

for example the preliminary serial driver is:

https://github.com/micro-os-plus/cmsis-plus/blob/xpack/include/cmsis-plus/drivers/serial.h

ReinhardKeil, 2016-09-27:
closing as there is no further activity.

closed 2016-09-27 by ReinhardKeil