ARM-software/CMSIS_4 #91: ARM_MATH_CM0_FAMILY and SIMD in DSP_Lib

sw, 2016-06-01:
Many DSP functions have SIMD optimizations, for example `arm_cfft_q15`. The optimized versions are selected by the absence of the `ARM_MATH_CM0_FAMILY` macro.

The macro `ARM_MATH_CM0_FAMILY` is defined in arm_math.h for Cortex-M0 processors.

But Cortex-M3 processors do not have the SIMD instructions. This leads to quite inefficient code on Cortex-M3 processors, because the SIMD instructions such as `__SHADD16` are emulated by the functions defined in arm_math.h

The lines
`#ifndef ARM_MATH_CM0_FAMILY`
should be changed to
`#if !defined (ARM_MATH_CM3) && !defined (ARM_MATH_CM0_FAMILY)`

I suspect this affects all modules there the existence of ARM_MATH_CM0_FAMILY is checked.

sw, 2016-06-01:
Also seems to affect the CMSIS_5 repository

GuentherMartin, 2017-01-11:
Selection is now done via macro ARM_MATH_DSP.
ARM_MATH_DSP specifies if DSP instructions should be used.
Please see https://github.com/ARM-software/CMSIS_5/commit/ec3081af4b42309566f1180dcb3f9db8508c7aa4.

ReinhardKeil, 2017-02-13:
released as part of CMSIS 5.0.1

closed 2017-02-13 by ReinhardKeil