ARM-software/CMSIS_4 #92: No const definitions for real input FFT instances

apullin, 2016-06-02:
There are no structure instances for the RFFT functions following the style of the CFFT structures. The downside to this is that if you are looping over several lengths, the compiler cannot select only the necessary twiddle factor tables. This can overrun flash for small devices quickly. Maybe in the case that an author is careful about declaring const values and the right optimizations are on.

I added definitions for the RFFT instances following the same pattern as the provided ones, and I am pretty sure that these are all defined in the same way that the init function would configure them at runtime:
https://github.com/apullin/CMSIS/tree/real_fft_const

The documentation in the "Real FFT" section on how to set these up is quite terse. For most end-user applications, input data is likely going to be all real-valued, so providing these instances should be useful.

Let me know if I should issue a pull request.

GuentherMartin, 2017-01-11:
Struct instances added to arm_const_structs.c.
Please see https://github.com/ARM-software/CMSIS_5/commit/ec3081af4b42309566f1180dcb3f9db8508c7aa4.

apullin, 2017-01-11:
Aw, I wanted to get a commit pulled into DSP-Lib, that would have been pro-status! Oh well.

ReinhardKeil, 2017-02-13:
released as part of CMSIS 5.0.1

closed 2017-02-13 by ReinhardKeil

apullin, 2018-01-23:
There does not appear to be declarations in arm_const_structs.h for the defined constant structures?