ARM-software/CMSIS_4 #95: Bug in arm_bitreversal_f32 when running on a Cortex M7 (stm32f7)

dmendesf, 2016-07-27:
When I call arm_cfft_f32 with the bitReverseFlag disabled program runs as expected on a STM32F745. When I enable the bitReverseFlag the  HardFault_Handler is activated inside the arm_bitreversal_f32 function. Don´t know exactly why, but did several tests and this is fully consistent (made a very small test program just to observe this behaviour). Will try soon with a STM32F4 to see what happens.

ReinhardKeil, 2017-02-13:
Does this problem still exist?

dmendesf, 2017-02-13:
No, it was a silly mistake. I don´t remember the details, but something 
like the wrong library being linked and things compiling but failing to 
work properly.

-- 
Daniel Mendes Fernandes
Gerente de Tecnologia
Tecnoclade Sistemas e Projetos
www.tecnoclade.com.br
  

Tel:  (21)3521-5216
Cel:  (21)99992-8668

O conteúdo desta mensagem e de seus anexos é de uso restrito
e confidencial, sendo o seu sigilo protegido por lei. Estas
informações não podem ser divulgadas sem prévia autorização
escrita. Se você não é o destinatário desta mensagem, ou o
responsável pela sua entrega, apague-a imediatamente e avise
ao remetente, respondendo a esta mensagem. A TECNOCLADE
não se responsabiliza por conclusões, opiniões, ou outras
informações nesta mensagem que não tenham sido emitidas
por seus integrantes.

ReinhardKeil, 2017-02-14:
OK, closing this issue then as it is sloved

closed 2017-02-14 by ReinhardKeil