ARM-software/CMSIS_4 #103: weird return value of __svcMutexWait/__svcMutexRelease

kyliuxing, 2016-12-14:
Target : RDA_5991H (cortex M4)

I feel the return of __svcMutexWait/__svcMutexRelease should be the value in enum list of osStatus, but sometimes it will return the value of mutex_id.
So my question is when the function will return the mutex_id as result?Is it normal?
It is a always issue when we use lwip stack to download and has blocked our developing, so please give some advices.
i reworked function to debug as below, osMutexRelease is same , please help check.

thank you!
RDA shileiyu

code:
lwip_sys_arch.c
sys_prot_t sys_arch_protect(void) {
    u32_t res = osMutexWait(lwip_sys_mutex, osWaitForever);
    if (res != osOK){
        printf("p addr = %p\r\n", lwip_sys_mutex);
        error("sys_arch_protect error, status = %x, addr = %p\n", res, &res);
        }
    return (sys_prot_t) 1;
}
rt_CMSIS.c
osStatus osMutexWait (osMutexId mutex_id, uint32_t millisec) {
osStatus res;
if (__get_IPSR() != 0U) {
return osErrorISR; // Not allowed in ISR
}
res = __svcMutexWait(mutex_id, millisec);
if(res != osOK)
printf("w %x, f %x\r\n", res, ((P_MUCB)mutex_id)->cb_type);
return res;
}

output:
w 1827f8, f 3
p addr = 001827f8
sys_arch_protect error, status = 1827f8, addr = 00184fa8

RobertRostohar, 2016-12-14:
__svcMutexWait/__svcMutexRelease should return values from enum osStatus and are implemented in that way (no other value is returned).

However those functions are actually called as service calls (SVC) and use stack for parameters and return value. Having not enough stack can lead to corruption including return values. This is probably the reason for the described behavior. Stack should be increased.

kyliuxing, 2016-12-14:
@RobertRostohar 
Thanks for your response, i have another question, __svcMutexWait/__svcMutexRelease just run svcMutexWait/svcMutexRelease in privilege mode through SVC calls, is it right?

RobertRostohar, 2016-12-14:
Correct. __svcXxx functions are calling the svcXxx functions through SVC (same parameters and return value).

kyliuxing, 2016-12-14:
I add some info before return in svcMutexWait/svcMutexRelease, but it wasn't printed, is it normal?
i feel it should be a 'w' or other info before "w 1827f8, f 3" if svcMutexWait has been called.
Thanks!

code:
osStatus svcMutexWait (osMutexId mutex_id, uint32_t millisec) {
  OS_ID     mut;
  OS_RESULT res;
  
  mut = rt_id2obj(mutex_id);
  if (mut == NULL) {
    mbed_error_printf("a\r\n");
    return osErrorParameter;
  }

  if (((P_MUCB)mut)->cb_type != MUCB) {
    mbed_error_printf("b\r\n");
    return osErrorParameter;
  }

  res = rt_mut_wait(mut, rt_ms2tick(millisec)); // Wait for Mutex

  if (res == OS_R_TMO) {
    mbed_error_printf("ccc\r\n");
    return ((millisec != 0U) ? osErrorTimeoutResource : osErrorResource);
  }
  mbed_error_printf("w\r\n");
  return osOK;
}
output:
r  //r is release ,w is wait
w
r
w 1827f8, f 3 //print in osMutexWait , please refer my first comment
p addr = 001827f8
sys_arch_protect error, status = 1827f8, addr = 00184fa8

RobertRostohar, 2016-12-15:
I cannot comment about mbed_error_printf() since I’m not familiar with its implementation.

However I would advise against adding printf style debugging to SVC functions since this might lead to failures (depends on the implementation of stdio retargeting). The new RTX5 is fully annotated with Events which can be used also for tracking return values (available on https://github.com/ARM-software/CMSIS_5/).

If you are really concerned about return values of svcXxx  functions then why not just debug using breakpoints.

But as already mentioned the problem is most likely in your configuration of stack size which results in __svcXxx functions returning wrong values despite that svcXxx functions return correct values.

Make sure that you have enough stack.

kyliuxing, 2016-12-15:
sorry, i didn't met the condition of debugging online~~~~
i'm not very clearly about how to enlarge the stack, when i create a thread, i pass the a parameter but i think it was used as PSP, Should i enlarge the MSP size? How to? 
Please kindly give some directions, Thank you!

RobertRostohar, 2016-12-15:
Threads are using PSP and the stack size is specified in the osThreadDef macro. When zero size is specified then the default stack size is used.

Default thread stack size (and others as main thread stack size) are configured in RTX configuration file (RTX_Conf_CM.c).

MSP is used for exceptions (including SVC) and the stack size is specified in the startup file.

kyliuxing, 2016-12-16:
@RobertRostohar  
I am not clear about how to set MSP's stack size, i found function set_stack_heap() in RTX_CM_lib.h, should i modify the ISR_STACK_SIZE in it?
freshman, so ~~~~sorry to trouble you
thank you!

RobertRostohar, 2016-12-16:
The code that you refer to is not part of this project but seems to be a modification done in ARMmbed/mbed-os project. Therefore you should check there and close this issue here,

kyliuxing, 2016-12-16:
Thank you for your help!

kyliuxing, 2017-01-06:
@RobertRostohar 
We analyze this issue more and found this is maybe a bug in mbed. SVC call would not be executed immediately, so if we use the return value at once, we may get wrong number form register R0, In this case, it is the mutex id. PLease help check it. thank you!

rt_cmsis.c
osStatus osMutexWait (osMutexId mutex_id, uint32_t millisec) {
  if (__get_IPSR() != 0U) {
    return osErrorISR;                          // Not allowed in ISR
  }
  return __svcMutexWait(mutex_id, millisec);
}
lwip_sys_arch.c
sys_prot_t sys_arch_protect(void) {
    if (osMutexWait(lwip_sys_mutex, osWaitForever) != osOK)
        error("sys_arch_protect error\n");
    return (sys_prot_t) 1;
}

RobertRostohar, 2017-01-11:
> We analyze this issue more and found this is maybe a bug in mbed.

Then it would be better to ask this in “mbed-os”.

Take another look at all the previous comments in this thread and also at answers in #105.

bulislaw, 2017-01-12:
@RobertRostohar we got a bug (https://github.com/ARMmbed/mbed-os/issues/3570) report from @kyliuxing following couple of issues across RTX and mbed-os, they report that this code doesn't work:
```
sys_prot_t sys_arch_protect(void) {
if (osMutexWait(lwip_sys_mutex, osWaitForever) != osOK)
error("sys_arch_protect error\n");
return (sys_prot_t) 1;
}
```
but when they add bunch of noops it's ok:
```
sys_prot_t sys_arch_protect(void) {
osStatus res;
res = osMutexWait(lwip_sys_mutex, osWaitForever);
__asm{
nop
nop
nop
nop
nop
nop
nop
nop
nop
nop
nop
nop
nop
nop
nop
}
if (res != osOK)
error("sys_arch_protect error\n");
return (sys_prot_t) 1;
}
``` 

They mention that 

> SVC call would not be executed immediately, so if we use the return value at once, we may get wrong number form register R0, In this case, it is the mutex id. PLease help check it. thank you!

What do you think about that? Is there a way `osMutexWait` can return in mentioned circumstances?

RobertRostohar, 2017-01-12:
@bulislaw 
No reasonable explanation by just looking at this limited information. Test reproducer might help ...

bulislaw, 2017-01-12:
Yes, @0xc0170  asked for one in https://github.com/ARMmbed/mbed-os/issues/3570 we should know better when we get it.

kyliuxing, 2017-01-12:
@RobertRostohar 
Could you give more explanation about my assumption? if the SVC call effectively later, why the  osMutexWait wouldn't return wrong value in R0? I feel my modification is influential in my HTTP test case, but seems not work well about ping test.(Please refer the following )

@0xc0170 
sorry, i'm not familiar with this issues system, i just want a place to discuss my problem, i will close 3570.
we met this issue in a ceaseless ping test and found the return value is mutex id, we also check the stack, like ISR, main and our own task's, but they don't seem like overflow.
We have the other case to reproduce, but i'm not sure it works to you because we need run our wifi protocol stack, maybe it has some effect.

reproduce step:
Connect a wrong ip address and send a HTTP request, it will be return -3011, then call recv can reproduce it.  I will attach my test code. 

code:
void  starttest()
{
_bpos = snprintf(_buffer, sizeof(_buffer) - 1, "GET %s HTTP/1.1\r\nHost: %s\r\nConnection: keep-alive\r\n Referer: %s\r\nUser-Agent: %s\r\nAccept-Encoding: %s\r\nAccept-Language: %s\r\n\r\n\n", path, HTTPS_SERVER_NAME, Referer, UserAgent, Encoding, Language);//some HTTP parameters
        
        SocketAddress address(HTTPS_SERVER_ADDR, HTTPS_SERVER_PORT);//wrong ip, port is 80
        _tcpsocket->set_timeout(1000);
        _tcpsocket->connect(address);

        _size = _tcpsocket->send(_buffer, strlen(_buffer) - 1);
        printf("send_size = %d\r\n", _size);// there should be -3011
        
        do {
            _size = _tcpsocket->recv((void *)RECV, RECV_BUFFER_SIZE);  
         }while(_size != 0);
        _tcpsocket->close();
        printf("recv over\r\n");
}

main
{
 //connect to internet, we use our own wifi protocol stack
starttest();
}

kyliuxing, 2017-01-16:
@RobertRostohar  @0xc0170 @bulislaw 
Does someone can give some help?

bulislaw, 2017-01-16:
The RDA_5991H doesn't seem to be supported by mbed-os, can you run mbed tests to have a feel how good your port is?

kyliuxing, 2017-01-16:
@bulislaw  our chip in developing and doesn't pass the certification by ARM, i don't know which test may reproduce this issue. could you recommend one?
@RobertRostohar  Could you kindly answer my question about SVC call?
thank you!

bulislaw, 2017-01-16:
Good start would be all rtos tests `"tests-mbedmicro-rtos-mbed-*"`

RobertRostohar, 2017-01-16:
@kyliuxing 

> if the SVC call effectively later, why the osMutexWait wouldn't return wrong value in R0? 

SVC call is not delayed. I have no other explanations for your case without looking at the whole project and preferably debugging it. A simple test reproducer might help.

kyliuxing, 2017-01-23:
@RobertRostohar 
We make a simulation and tried to find what happened.
We have two threads, thread1 and thread2, thread1 will call osThreadYield. this is the precondition.
Thread2 will call osMutexRelease to release mutex, but before the SVC instruction, we met a systick, then pendSV to switch to thread1. Instantly, thread1 call SVC handler to run svcThreadYield and switch back to thread2. But the return point is blow the SVC instruction, so the svcMutexRelease didn't executed actually.
We try to remove the osThreadYield in thread1, the problem wasn't reproduced.
could you give some help? 
By the way, is it convenient discussed in email? i can attach more info.
Thank you!

RobertRostohar, 2017-01-23:
Check if the interrupt priorities are correctly configured.

RTX uses SVC, PendSV and SysTick interrupts which needs to be properly configured:
- pre-emption priority of SysTick/PendSV is lover than SVC
- priority grouping (PRIGROUP): at least 1-bit is used for pre-emption

This is configured automatically during osKernelInitialize. However if user changes the mentioned priorities than the above requirements need to be respected.

And again check if there is enough stack (MSP/PSP).

kyliuxing, 2017-01-24:
thanks for your suggestion and it help to resolve the problem!
we set the same priority of SysTick/PendSV and SVC,  when systick happened before one cycle of the SVC, CPU will run systick firstly and switch PSP,  so the SVC get the wrong PSP.

thanks again!

closed 2017-01-24 by kyliuxing