ARM-software/CMSIS_4 #105: Some question about svcMutexWait

kyliuxing, 2017-01-05:
I write a test case but the result confuse me.
1. If i use osWaitForever in svcMutexWait, when it will return osErrorTimeoutResource? if some one has hold the mutex, i feel thread will be blocked in rt_mut_wait() and won't return until some one release it.
2. When i see trace "w3 c1 id 001035c4 level 1", why the "aa res 0" wasn't printed? Is svcMutexWait() should be return osErrorTimeoutResource  or osErrorResource to osMutexWait()?
Thank you!

output:
appleowner
res = 0
aa res 0
appleowner w res 0
applehunger
res = 1
w3 c1 id 001035c4 level 1
appleowner r res 0
aa res 0
applehunger res 0

test case:  
main.c
void appleowner(void *arg)
{
	osStatus res;
	printf("appleowner\r\n");
	res = osMutexWait(apple_id, osWaitForever);
	printf("appleowner w res %d\r\n", res);
	osDelay(50);
	res = osMutexRelease(apple_id);
	printf("appleowner r res %d\r\n", res);
	osDelay(50);
	while(1);
}
void applehunger(void *arg)
{
	osStatus res;
	printf("applehunger\r\n");
	osDelay(5);
	//while(1){
	res = osMutexWait(apple_id, osWaitForever);
	printf("applehunger res %d\r\n", res);
	//}
	while(1);
}
rt_CMSIS.c
svcMutexWait()
{
....................
 res = rt_mut_wait(mut, rt_ms2tick(millisec)); // Wait for Mutex
  mbed_error_printf("res = %x\r\n", res);
  if (res == OS_R_TMO) {
    mbed_error_printf("w3 %x id %p level %x\r\n", osErrorTimeoutResource, mutex_id, ((P_MUCB)mut)->level);
    return ((millisec != 0U) ? osErrorTimeoutResource : osErrorResource);
  }

  return osOK;
}
osStatus osMutexWait (osMutexId mutex_id, uint32_t millisec) {
  osStatus res;
  if (__get_IPSR() != 0U) {
    return osErrorISR;                          // Not allowed in ISR
  }
  res = __svcMutexWait(mutex_id, millisec);
  printf("aa res %x\r\n", res);
  return res;
}

RobertRostohar, 2017-01-11:
> If i use osWaitForever in svcMutexWait, when it will return osErrorTimeoutResource?

Never. If mutex cannot be acquired then the thread will block until another thread releases that mutex. At that point the function will return but with osOK.

This is fundamental behavior. It should not be necessary to dig into RTOS implementation itself and try to understand the details (requires deep knowledge).

> if some one has hold the mutex, i feel thread will be blocked in rt_mut_wait() and won't return until some one release it.

Correct (expect it will be blocked in svcMutexWait).

> When i see trace "w3 c1 id 001035c4 level 1", why the "aa res 0" wasn't printed?

Because the thread was blocked inside svcMutexWait and execution continued with another thread (see below).

The test case works as expected. 

Execution flow of the test is the following:
-	appleowner thread starts (output: appleowner)
-	appleowner thread acquires mutex (output: res = 0, aa res 0)
-	appleowner thread outputs: appleowner w res 0
-	appleowner thread goes into delay
-	execution switches to applehunger thread
-	applehunger thread outputs: applehunger
-	applehunger thread tries to acquire mutex without timeout but fails and is blocked (output: res = 1, w3 c1 id 001035c4 level 1)
-	return value of svcMutexWait is set to osErrorTimeoutResource, however the function does not return at this point since the thread is blocked
-	execution switches to appleowner thread
-	appleowner thread releases the mutex (output: appleowner r res 0)
-	return value of blocked svcMutexWait is changed to 0 by the system
-	appleowner thread goes into delay
-	execution switches to applehunger thread inside osMutexWait (after svcMutexWait), output: aa res 0, applehunger res 0

kyliuxing, 2017-01-12:
@RobertRostohar 
Thank you for nice answer! it is very help!
But i still confused about when the thread will be switched, i feel it should be happened in rt_mut_wait() and would not print msg like "res = %x\r\n" and "w3 %x id %p level %x\....",  the switch will wait the func completed?
thank you!

RobertRostohar, 2017-01-12:
> when the thread will be switched, i feel it should be happened in rt_mut_wait() 

No. Thread switching is implemented in the SVC handler.

Example:
__svcMutexWait calls the SVC handler which calls svcMutexWait and returns back to the handler. Then if a thread switch is required this happens at the end of the SVC handler.

kyliuxing, 2017-01-12:
I'm clear. Thank you for help! have a good day!

closed 2017-01-12 by kyliuxing